prusa3d/PrusaSlicer #13559: Brim is missing on inner side

aredspider, 2024-11-05:
### Description of the bug

When I enable brim, it is missing in the inner side of a closed object. It seems like this happens only when I have more than one instance:
[image]

### Project file & How to reproduce

Open PrusaSlicer, add model, select outer and inner brim, slice
[duplo-brio-adapter-remix_v5.zip](https://github.com/user-attachments/files/17629499/duplo-brio-adapter-remix_v5.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1+linux+x64+GTK3

### Operating system

Ubuntu 24.04.1 LTS

### Printer model

K2Pro

u89djt, 2024-11-06:
(Fellow user taking an interest and fleshing out some observations)
https://www.printables.com/model/397173-duplo-brio-adapter-remix 

I guess the problem lies in an interaction between; the hierarchical representation of objects and regions on the bed, and the implementation of boundary definition and crossing counting.

Workaround 1: set them as separated objects.
[image]
[image]
But you don't get a brim on the outside of the inner cylinder.

Thinking about the problem a bit more to see if there could be any inspiration for a solution:
It's a failure to provide brims for all meeting points between object and bed.
The instruction is to provide brims to the inside and outside of objects.
The brim to the outside of the inner cylinder could be called inside for the main object, so there's conceptual contention there.
We can note - but shouldn't reflexively interpret - the presence of the unambiguously internal brim in the single instance case.

Inspired by recognising the contention for the internal/external brim:
Workaround 2, which gets you a brim inside and outside the inner cylinder wall.
Rebuild the object to separate the inner cylinder as its own mesh to present to the slicer as a separate body.
Constructed in Fusion from
[image]
to
[image]
(Taking the opportunity to modify the slope of the top of the fillet in the tongue to avoid bridge extrusions.)
To get the slice you want, replace the original mesh with the new pair of meshes,
[image]
don't panic
[image]
Establish number of copies and layout, then split to objects, not to parts. If you split to parts, it doesn't work.
If you need to rearrange things on the plate afterwards, you'll have to undo the separation first.
That appears to present the slicer with unambiguous insides and outsides.
Verifying that the object extrusions are the same as the original:
[image]
Original model on the left, split model on the right.

Here's the bed before and after splitting to objects.
[brimshack.zip](https://github.com/user-attachments/files/17647010/brimshack.zip)

u89djt, 2024-11-06:
Moving the objects apart so the outer brims between objects don't touch doesn't change things.

Snuff1eupagus, 2024-11-07:
@aredspider

This is easily overcome in PrusaSlicer simply make an Object, A cylinder, simply place the cylinder inside one of the instances apply Skirt and Brim settings to the new cylinder Object. It will inherent the settings you have already provided from the global brim settings, and your done. No need to break anything apart or change your model, just add the Object add the Skirt & Brim settings and your done. 

Here's an example file for you. 

[duplo-brio-adapter-remix_v5_Snuff_.zip](https://github.com/user-attachments/files/17657984/duplo-brio-adapter-remix_v5_Snuff_.zip)

By the way I reduced the amount of Brim separation gap: from 0.3 to 0.07, change it back if you must, but 0.3 is a lot,

and here's a video that might help you to understand how this works

https://www.youtube.com/watch?v=0Zxl1WOJm90

Regards

aredspider, 2024-11-07:
@Snuff1eupagus @u89djt Thanks for the ideas, will try these workarounds. This is meant to be a bug report, though

u89djt, 2024-11-07:
Workarounds solve problems until the bug is fixed :)

Snuff1eupagus, 2024-11-07:
In this case there is no actual bug, this is simply a lack of understanding of what PrusaSlicer is capable of. The brims are doing what the brims are supposed to do, the control is working properly, as a user you are expected to educate yourself and understand the methods. 
The information has been provided and it is clear.
BTW I'm just a user not a Prusa representative.

aredspider, 2024-11-08:
@Snuff1eupagus no, I don't see it that way. The results differ, if you have 1 or 2 instances of the same object. Read the bug description. They should be identical.

u89djt, 2024-11-08:
@Snuff1eupagus This is a bug. A single object with active brims yields one pattern, but another when appearing as multiple instances. The user should not expect that behaviour based on any of the information available. This information is available in the bug report.
Your inheritance hack is interesting. It restores the brim while retaining the ability to play with the settings of the ensemble as single modifiers.
Presumably it would be worth anchoring the dot to the brim.
[image]
You said "this is in no way a workaround" That is incorrect. It's a workaround in this case.
You said "most people simply don't know how to use PrusaSlicer and..." That would be insulting if it was accurate. They did exactly the right things and the bug caused a problem.

Snuff1eupagus, 2024-11-10:
I'm sorry that you may have felt some insult, that most serenely was not my intent, However I feel that you are not understanding what slicer is capable of and how brims can be utilized. My instruction doesn't require that you have a full understanding, however you should strive to understand what is possible, how it works, before stating that there is a bug present and I don't see that evident in either of your statements. In fact I can show you if your wiling to learn that both your statements are untrue, even by  your standards. but you have to be willing to grow.  When I'm on the Prusa Forums I recognize that persons are total newbies.
When I'm on Github I suspect that persons are in some form or fashion either sent here by newbies, or they are programmers that can't think out of the box , I see a great deal of this here, and I'm sad for them. But I will not give to those sad ways of thinking they know, There are persons in the world that are beyond these ways of thinking and most certainly know better, are you willing to be one of them or are you a guesser. so far your a guesser in my book. Is that an insult? that's your mind that decides that, or are you smarter then that? You make that decision.  Sorry if you can't see it. For sure there is a movent beyond your chosen GitHub experience.

Snuff1eupagus, 2024-11-10:
[u89djt] I'm truly sorry that you cant understand what has been feed to you. what the software can truly be depends on users understanding what it can and needs to be, you have shown that you are not that person. 
please try to move forward and see what things can be and think differently. 

IT MATTERS.

u89djt, 2024-11-10:
Noting in the report "It seems like this happens only when I have more than one instance", open the project file and hit the minus icon twice to remove the additional instances, and note that the brim looks like this:
[image]
Until it looks the same with multiple instances, the user needs a workaround.

Snuff1eupagus, 2024-11-10:
LEARN AND GROW

aredspider, 2024-11-10:
😆 Haha, okay. Who is Neophyl, btw?

Well, moving on to a more serious conversation: I would suspect the error to be somewhere in the [inner brim for loop](https://github.com/prusa3d/PrusaSlicer/blob/5dc04b4e8f14f65bbcc5377d62cad3e86c2aea36/src/libslic3r/Brim.cpp#L337) where the instances are iterated, maybe L349? I didn't have the time to spin up the debugger, yet.

It seems to be a general issue, when an object has 'two' insides.
For example, this also happens when you have two 'O' letters inside of each other.
Single instance:
[image]
Two instances:
[image]

Snuff1eupagus, 2024-11-10:
Oh my, I must have drank to much last night, I don't even remember writing this. 😆

deckar01, 2026-09-24:
I was able to reproduce this 3.0.0-alpha12. Notching the outer perimeter causes the slicer to brim the inner face, but the brim is extended to the inside of the outer perimeter. Makes me think this needs a paint-like control.

[image] notch hack -> [image]