prusa3d/PrusaSlicer #13560: undefined reference to `prusaslicer_png_destroy_read_struct' in compiling

dariosharp, 2024-11-05:
### Description of the bug

Hi, 

I'm trying to compile PrusaSlicer from the sourcecode, but I get an error during the compilation. I have followed the building guide and I have properly compiled the dipendeces but when I try to compile the whole program I get the following error:

```
/usr/bin/ld: libslic3r/liblibslic3r.a(PNGReadWrite.cpp.o): in function `Slic3r::png::png_read_callback(png_struct_def*, unsigned char*, unsigned long)':
PNGReadWrite.cpp:(.text+0xb): undefined reference to `prusaslicer_png_get_io_ptr'
/usr/bin/ld: libslic3r/liblibslic3r.a(PNGReadWrite.cpp.o): in function `Slic3r::png::is_png(Slic3r::png::ReadBuf const&)':
PNGReadWrite.cpp:(.text+0x6f): undefined reference to `prusaslicer_png_sig_cmp'
/usr/bin/ld: libslic3r/liblibslic3r.a(PNGReadWrite.cpp.o): in function `Slic3r::png::decode_png(Slic3r::png::IStream&, Slic3r::png::Image<unsigned char>&)':
PNGReadWrite.cpp:(.text+0xc1): undefined reference to `prusaslicer_png_sig_cmp'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x106): undefined reference to `prusaslicer_png_create_read_struct'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x11b): undefined reference to `prusaslicer_png_create_info_struct'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x13c): undefined reference to `prusaslicer_png_set_read_fn'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x14a): undefined reference to `prusaslicer_png_set_sig_bytes'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x158): undefined reference to `prusaslicer_png_read_info'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x166): undefined reference to `prusaslicer_png_get_image_width'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x17a): undefined reference to `prusaslicer_png_get_image_height'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x18e): undefined reference to `prusaslicer_png_get_color_type'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x19e): undefined reference to `prusaslicer_png_get_bit_depth'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x202): undefined reference to `prusaslicer_png_read_row'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x23e): undefined reference to `prusaslicer_png_destroy_info_struct'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x255): undefined reference to `prusaslicer_png_destroy_read_struct'
/usr/bin/ld: libslic3r/liblibslic3r.a(PNGReadWrite.cpp.o): in function `Slic3r::png::write_rgb_or_gray_to_file(char const*, unsigned long, unsigned long, int, unsigned char const*)':
PNGReadWrite.cpp:(.text+0x2f6): undefined reference to `prusaslicer_png_create_write_struct'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x30f): undefined reference to `prusaslicer_png_create_info_struct'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x339): undefined reference to `prusaslicer_png_set_longjmp_fn'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x38f): undefined reference to `prusaslicer_png_set_IHDR'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x3a8): undefined reference to `prusaslicer_png_malloc'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x3dc): undefined reference to `prusaslicer_png_malloc'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x40e): undefined reference to `prusaslicer_png_init_io'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x426): undefined reference to `prusaslicer_png_set_rows'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x43f): undefined reference to `prusaslicer_png_write_png'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x461): undefined reference to `prusaslicer_png_free'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x476): undefined reference to `prusaslicer_png_free'
/usr/bin/ld: PNGReadWrite.cpp:(.text+0x490): undefined reference to `prusaslicer_png_destroy_write_struct' 
/usr/bin/ld: libslic3r/liblibslic3r.a(PNGReadWrite.cpp.o): in function `Slic3r::png::decode_png(Slic3r::png::IStream&, Slic3r::png::Image<unsigned char>&) [clone .cold]':
PNGReadWrite.cpp:(.text.unlikely+0x17): undefined reference to `prusaslicer_png_destroy_info_struct'
/usr/bin/ld: PNGReadWrite.cpp:(.text.unlikely+0x2a): undefined reference to `prusaslicer_png_destroy_read_struct'
/usr/bin/ld: /mnt/work/dbertani/PrusaSlicer-version_2.8.1/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.2.a(imagpng.cpp.o): in function `wx_PNG_stream_reader':
imagpng.cpp:(.text+0x1d): undefined reference to `prusaslicer_png_get_io_ptr'
/usr/bin/ld: /mnt/work/dbertani/PrusaSlicer-version_2.8.1/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.2.a(imagpng.cpp.o): in function `wx_PNG_stream_writer':
imagpng.cpp:(.text+0x4d): undefined reference to `prusaslicer_png_get_io_ptr'
/usr/bin/ld: /mnt/work/dbertani/PrusaSlicer-version_2.8.1/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.2.a(imagpng.cpp.o): in function `wx_PNG_warning':
imagpng.cpp:(.text+0x286): undefined reference to `prusaslicer_png_get_io_ptr'
/usr/bin/ld: /mnt/work/dbertani/PrusaSlicer-version_2.8.1/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.2.a(imagpng.cpp.o): in function `wx_PNG_error':
imagpng.cpp:(.text+0x8cf): undefined reference to `prusaslicer_png_get_io_ptr'
/usr/bin/ld: /mnt/work/dbertani/PrusaSlicer-version_2.8.1/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.2.a(imagpng.cpp.o): in function `wxPNGHandler::GetLibraryVersionInfo()':
imagpng.cpp:(.text+0x1397): undefined reference to `prusaslicer_png_get_header_version'
/usr/bin/ld: /mnt/work/dbertani/PrusaSlicer-version_2.8.1/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.2.a(imagpng.cpp.o): in function `wxPNGHandler::SaveFile(wxImage*, wxOutputStream&, bool)':
imagpng.cpp:(.text+0x2536): undefined reference to `prusaslicer_png_create_write_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x254e): undefined reference to `prusaslicer_png_create_info_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x2586): undefined reference to `prusaslicer_png_destroy_write_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x2e16): undefined reference to `prusaslicer_png_set_write_fn'
/usr/bin/ld: imagpng.cpp:(.text+0x3c80): undefined reference to `prusaslicer_png_set_filter'
/usr/bin/ld: imagpng.cpp:(.text+0x413f): undefined reference to `prusaslicer_png_set_compression_level'
/usr/bin/ld: imagpng.cpp:(.text+0x45fe): undefined reference to `prusaslicer_png_set_compression_mem_level'
/usr/bin/ld: imagpng.cpp:(.text+0x4946): undefined reference to `prusaslicer_png_set_compression_strategy' 
/usr/bin/ld: imagpng.cpp:(.text+0x4cc2): undefined reference to `prusaslicer_png_set_IHDR'
/usr/bin/ld: imagpng.cpp:(.text+0x4d9a): undefined reference to `prusaslicer_png_set_sBIT'
/usr/bin/ld: imagpng.cpp:(.text+0x4dad): undefined reference to `prusaslicer_png_write_info'
/usr/bin/ld: imagpng.cpp:(.text+0x4dc0): undefined reference to `prusaslicer_png_set_shift'
/usr/bin/ld: imagpng.cpp:(.text+0x4dcc): undefined reference to `prusaslicer_png_set_packing'
/usr/bin/ld: imagpng.cpp:(.text+0x5159): undefined reference to `prusaslicer_png_write_rows'
/usr/bin/ld: imagpng.cpp:(.text+0x5198): undefined reference to `prusaslicer_png_write_end'
/usr/bin/ld: imagpng.cpp:(.text+0x51ab): undefined reference to `prusaslicer_png_destroy_write_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x57af): undefined reference to `prusaslicer_png_set_PLTE'
/usr/bin/ld: imagpng.cpp:(.text+0x57f8): undefined reference to `prusaslicer_png_set_tRNS'
[... 31 lines cut ...]
make: *** [Makefile:146: all] Error 2
```

Can you help me?

### Project file & How to reproduce

none

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer version 2.8.1

### Operating system

Debian GNU/Linux trixie

### Printer model

none

nmschulte, 2024-11-21:
For some reason, the build/configuration is trying to link to the system `libpng` on my machine.  I don't know the correct fix to the CMake bits, but I was able to workaround the issue by editing the generated `link.txt`:
- **`prusa-slicer`**: `build-version_2.9.0-alpha1/src/CMakeFiles/PrusaSlicer.dir/link.txt`
- **`arrange_tests`**: `build-version_2.9.0-alpha1/tests/arrange/CMakeFiles/arrange_tests.dir/link.txt`

change the PNG `.so` references, after `libgmp.a`, to the built `libpng.a` (sibling to `libgmp.a`), run `make`

themanyone, 2024-11-28:
I think that happens when running the `cmake .. -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF` 
without  `-DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local` due to line wrapping in the cut and paste.

dariosharp, 2024-11-28:
Actually the command that I run is:
`cmake .. -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF -DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local`

I followed the guide to build the software using the release source code PrusaSlicer 2.8.1.

I have not already tried to edit the link.txt file. I'm gonna do it as soon as possible.

bkerler, 2024-12-08:
I can confirm to have the same issue on debian 12 as @dariosharp. It seems to use the system based libpng16 path instead of the built deps libpng16, causing the linkage error on cmake version 3.30.5.

marcbres, 2024-12-21:
I can confirm the same on my debian testing installation. After following the instructions by @nmschulte and executing a make clean before the final make, I've been able to compile the prusa-slicer. 
Thank you all!

dariosharp, 2024-12-24:
I have switched on ubuntu 24.04 (with cmake 3.28.3) and now I'm able to properly compile the software.

closed 2026-09-29 by prusaslicer-bot[bot]