prusa3d/PrusaSlicer #13563: Solid Fill pattern options

fabianmuehlberger, 2024-11-06:
**Is your feature request related to a problem? Please describe.**
It was described that on short fill lines the pattern is automatically changed. 

As Godrak said: 
https://github.com/prusa3d/PrusaSlicer/issues/10677#issuecomment-1569828928

The Solid Fill pattern was noticed here:
https://github.com/prusa3d/PrusaSlicer/issues/11207
https://github.com/prusa3d/PrusaSlicer/issues/10677
https://github.com/prusa3d/PrusaSlicer/issues/10573

**Describe the solution you'd like**
Allow users to choose the Solid fill pattern from a drowpdown similar to top or bottom pattern. Options I would like: 
- Concentric, aligned rectilinear,

**Describe how it would work**
- Have an UI element to allow changing the solid fill pattern. 

**Additional context**
- While printing concentric parts and optimizing the slicer, I noticed that a concentric solid fill pattern would be beneficial for printing tubes.

fabianmuehlberger, 2024-11-06:
For clarificartion: 
this is how the fill pattern currently looks: 
[image]

advantageous would be a circular pattern, same as the first layer:
[image]

Snuff1eupagus, 2024-11-07:
u89djt

I'm not 100% sure I'm following what your saying, however if you wish to have what amounts to concentric Fill for the entire model.
You would first set: 

Fill pattern: Concentric
Fill density: 100%
Then
Solid layers:
Top: 1
Bottom: 1
and
Minimum shell thickness: 
Top: 0
Bottom: 0

This will make all layers print as Concentric loops.

No reason what so ever for a new feature to get this outcome, this has been with us for years, and is easy to produce, development time can be better spent on more worthy features. Examples of features available upon request.

Regards

u89djt, 2024-11-07:
@Snuff1eupagus I made a bad job of quickly saying that :) And trying to illuminate around the common case (?) where one changes to 100% infill and the slicer suggests only one of the compatible fill patterns, since they want a choice of patterns.
I then wondered if concentric surface fill or a variant might succeed over sparse fill alone. The linked issues speak to judder, so minimizing the frequency and number of direction changes seems to be a goal in there.
[concentrictop.zip](https://github.com/user-attachments/files/17659494/concentrictop.zip)
[image]

fabianmuehlberger, 2024-11-07:
My feature request was not to have a 100% concentric infill. 
I still want to be able to choose my regular infill as needed. 

It was, to have an option to change the Solid infill (this is not the controlable infill but the horizontal layers after the first layer and before the top)

Why am I asking for this: 
Printing concentric parts with PETG-CF. On short infill lines, there is high buildup of filament on the nozzle, leading to failed prints. I noticed that this is not the case with long continous filling. 
Therefore, having control over the fill pattern solves a problem with print qualtiy, as others pointed out that the slicer does this by its own to prevent short strokes, it is a step further for letting me control the behavior of the print. 

A different solutions woud be: 
- have a controlable amount of bottom an top layers. 
- an option to make the solid infill with the same fill pattern as bottom and top fill

u89djt, 2024-11-07:
@fabianmuehlberger post a project file or three where you want this? You need to zip them. If you want dev time applied, that will benefit from concrete motivational examples, some simple, some complex.

Snuff1eupagus, 2024-11-07:
I misunderstood the request.

u89djt, 2024-11-08:
For the sake of curiosity: this might be overlap/speed/cooling away from success.
[concentrictop.zip](https://github.com/user-attachments/files/17678405/concentrictop.zip)
[image]
[image]

fabianmuehlberger, 2024-11-13:
Hey @Snuff1eupagus and @u89djt. I prepared a small example case for the concentric fill pattern. 
This  applies to tubes, or parts that have walls more an a few mm thickness, and open bottom and top.  In this case, the rectilinear fill pattern creates short print lines, that are not beneficial. 

- first layer of the print
[image]
- second layer of the print
[image]

Zip with the project and STEP files

[concentric_fill.zip](https://github.com/user-attachments/files/17733322/concentric_fill.zip)

fabianmuehlberger, 2024-11-14:
A short update: 
I found the modifier option to add a volume with custom settings to the part. This works, although it takes some time to adjust the layers correctly, it delivers the desired outcome.
[image]

For me, this is not an ideal but a sufficient solution. If you decide not to follow this feature request.

localJetson, 2025-02-28:
@u89djt @Snuff1eupagus 
I hope I'm not too late to comment here and I hope I'm not breaking community rules by comparing.

I understand that there can be workarounds with how flexible Prusa is but something this integral that would apply to a very large number of parts should be a valid feature request.

I believe the issue the OP is referring to is extremely fair and feels like an oversight. The fill pattern we are talking about is specific to the bottom and top layer of prints where, by design and requirements in most prints, they are printed at 100% density. I have parts I really only print at 15% but I still heavily need top and bottom layers to be 100% to properly seal the part. 

The problem comes about because of part geometry. This infill pattern, although it guarantees the 100% density requirements, produces other problems with the multiple stops and restarts as well as lots of small movements that generate vibrations and unnecessary forces on the part and printer. My main bother is about the poor print quality and the slight unevenness through the entire infill due to all the seams. It also just takes longer by a lot with little to no real benefit.

It results to me so frustrating since there is an option that changes the fill pattern but just for the outside layers with no way of expanding it to the rest of the top and bottom layer. Given it already does the work, it seems like it could be an easy fix and if anything, just less work for the slicer to need to change the infill pattern.

Cura slicer lets you change these parameters and allows you to fine tune the resulting gcode for these layers so I believe it should be implemented on Prusa as well.

guiweber, 2025-09-27:
It would be a really nice feature to have. The workarounds are rather convoluted for what is a common and simple requirement for tubular parts.

Also seems to be related to #2147 and #5803

guiweber, 2026-09-18:
Still relevant as of 3.0 alpha 11

Jan-Soustruznik, 2026-10-06:
Duplicate of #2147

closed 2026-10-06 by Jan-Soustruznik

guiweber, 2026-10-09:
FYI - The issue this was a duplicate of (#2147) has been closed and moved to a discussion here: https://github.com/prusa3d/PrusaSlicer/discussions/15999

Those who have examples of where this feature would be useful should go over there and repost them. For some reason the discussion is labeled "very unusual idea", so it would be good to show how "usual" the usefulness of this feature would be.