prusa3d/PrusaSlicer #13564: Print sequence: By layer or By object

inovatorius, 2024-11-06:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
OrcaSlicer has a Special Mode, where Print sequence can be defined by layer or by object.
[image]

**Describe how it would work**
By layer method it is what used know in Prusa Slicer right now. You never know what next object or part will be printed as this slicing method lives a life on its own, however several things that adds randomness to the slicing process are:

1. Object or part rotation/angle - every time you rotate the part or object you get different order of printing sequence.
2. Order of object in the the list - whenever you move object up or down in the list, the printing sequence changes.

For example, this is the sequence of objects I have:
[image]

Because why not?

**Describe alternatives you've considered**
The only alternative to this randomness is to spend another 30 minutes to find the ideal combination of object angles, their position in the list and pray to Lord and Savior Josef Prusa.

**Additional context**
Prusa Slicer should obey.

inovatorius, 2025-01-31:
Is this hardcore or we need a crowdfunding?

neophyl, 2025-01-31:
In Prusa slicer 'Print Sequence' is called Complete Individual Objects.  The setting is located in Print Settings>Output Options>Sequential Printing>Complete individual objects.  Its a yellow coded setting so not available in beginner mode.  Its been in there forever.  Its not something new added to Orca.
Orca's Intra-Layer order is an extra function though that PS does not as yet have.

If you go to the plater with multiple objects and press 'E' to bring up the labels you will see that the print sequence is shown

[image]
It follows the order of the objects in the right hand list.

closed 2026-09-29 by prusaslicer-bot[bot]