prusa3d/PrusaSlicer #13565: Filament "DasFilament PETG" doesn't appear in the list

pepouille, 2024-11-07:
### Description of the bug

When I select the various filament profiles I want to be included, the filament from the mark "DasFilament PETG" never appears in the dropdown list of available filaments; the other one from "DasFilament" (DasFilament PLA) appear well.

### Project file & How to reproduce

To choose it to be setup:
Configuration-> Configuration Wizard -> Filaments (in the list) -> Printer (All) -> Type (PETG) -> Vendor (Das Filament) -> profile (Das Filament PETG *)

To select (but not possible due to the filled bug!) :+1: 
In plater view -> Dropdown Filaments lits on the right panel --> Das filament PETG is not shown

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.1+linux-x64-GTK3

### Operating system

Debian GNU/Linux 12 (bookworm)

### Printer model

Prusa MK4

rtyr, 2024-11-07:
> Das Filament PETG *

Such filament profile does not exist within Prusa vendor space (for Prusa printers) at the moment. If you see it in the wizard, it is likely because you are also using printer from other vendor, which have this profile available. This is also indicated by * mark (it says the profile is only compatible with some printers from your list).

neophyl, 2024-11-07:
That just means its not officially been tested so is not available for your selected printer.  Install it for another printer and then modify it to remove the printer requirements and save it as a custom filament.  That way it will then show up for your MK4.
Prusa cant test every single filament on every single model.  Also make sure that in Preferences that you have Show Incompatible print and filament presets ticked
[image]

rtyr, 2024-11-07:
I would suggest to use some Prusa filament profile of the same type as a base (for example Generic PETG) and use profile meant for other printer vendor only as an inspiration for changes. There are some specific important parameters, which are not present in filament profiles for other vendors (like pressure advance values, MMU parameters, ramping z-hop settings, retraction overrides, etc.).