prusa3d/PrusaSlicer #13566: Multiple Wipe towers

Pete1001, 2024-11-07:
## Feature Request: Enhanced Wipe Tower Management for Multi-Material Printing

### Is your feature request related to a problem? Please describe.
This feature request addresses a recurring issue encountered during multi-material printing, specifically when using a combination of BVOH and PETG, or PLA and PETG. The primary issue is that the wipe tower becomes detached from the build plate, leading to compromised print quality and frequent print failures. This problem occurs consistently on the Prusa XL printer. Despite multiple interactions with support, I have not received a resolution or effective guidance for mitigating this issue.

### Describe the Solution You'd Like
I am requesting the capability to select and assign multiple wipe towers for each specific type of filament being used in a multi-material print. For example:

- When printing with PLA and PETG, the ability to assign two separate wipe towers, one dedicated to each filament type, would help maintain print stability and prevent cross-contamination.
- Similarly, if I add an additional filament like BVOH or PVA to the print, I would like to designate a unique wipe tower for each material, allowing for isolated material management and improved print reliability.

### Describe How It Would Work
The feature would enable users to configure multiple wipe towers directly in the slicer settings. For each filament or material selected, an additional wipe tower could be assigned, providing a tailored approach to multi-material printing that ensures better adhesion and minimizes contamination between filaments.

### Describe Alternatives You've Considered
To resolve this print failure issue, I have explored numerous alternatives, including:

- Experimenting with different filaments and print head configurations
- Using thoroughly dried filaments
- Replacing extruders and testing various hardware adjustments
- Modifying hundreds of slicer settings based on support recommendations

Unfortunately, none of these attempts have successfully addressed the problem, underscoring the necessity for a multi-wipe tower feature to ensure consistent, high-quality prints in multi-material setups.

### Additional Context
Please find any additional context, diagrams, or sketches illustrating this feature request below, as they may provide further insight into how this enhancement could resolve the issue effectively.

[image]

u89djt, 2024-11-07:
Definitely needed. Previous report https://github.com/prusa3d/PrusaSlicer/issues/13161
If we could assert the order of printing of objects, I guess it would be straightforward to design a wipe object for each extruder on the bed ourselves.

tomecp, 2024-11-17:
This would help and solve a lot of problems with "multi-nozzle size" prints as well!

KlausKraemer, 2025-02-18:
This is abolutely needed to make the XL fully functional. A shame, Prusa did not develop a solution yet!

Jan-Soustruznik, 2025-04-14:
similar of #13161