prusa3d/PrusaSlicer #13572: ошибка

kaszbk, 2024-11-09:
### Description of the bug

При закрытии программы PrusaSlicer 2.8.1, вылазит ошибка,В чем может быть причина? 

### Project file & How to reproduce

[Снимок экрана 2024-11-09 111322.zip

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

 PrusaSlicer 2.8.1

### Operating system

windows 10_64b

### Printer model

prusa

u89djt, 2024-11-09:
Скриншот и файл проекта не видны.
The screenshot and project file are not visible.

kaszbk, 2024-11-09:
----------------

u89djt, 2024-11-09:
Все еще ничего. Посмотрите на сайте...

rtyr, 2024-11-10:
You have to use the web interface to upload screenshot. It is not possible send it via e-mail attachement. Ale please write in English.

kaszbk, 2024-11-11:
Deleted the folder c:\Users \*\AppData\Roaming\PrusaSlicer\,deleted PrusaSlicer 2.8.1, cleaned the registry. I installed PrusaSlicer 2.8.1, the error remained. There was no such error in the version of PrusaSlicer 2.7.1.

closed 2026-09-29 by prusaslicer-bot[bot]