prusa3d/PrusaSlicer #13573: Extrusion Width Miscalculation and Flow::spacing() Error with Nozzle Diameter Matching Layer Height

adamrumjahn, 2024-11-09:
### Description of the bug

PrusaSlicer Version: 2.8.1

Operating System: Windows 10

Perimeter Generator: Arachne (note, this error only occurs with Arachne, not Classic, so I think its related to this)

Printer Model: ORION A150 (custom profile), but I also reproduced this error switching to the Prusa MK3 profile as well

Issue Description: I'm encountering a persistent issue in PrusaSlicer when slicing a model with the following settings:

Nozzle Diameter: 0.40 mm
Layer Height: 0.30 mm
Default Extrusion Width: 100% (intended to match the nozzle diameter)
When I attempt to slice the file with these settings, I receive the following errors:

"ERROR: extrusion_width=0.3 mm is too low to be printable at a layer height 0.3 mm" – This happens even though the extrusion width should automatically set to 0.4 mm, which is greater than the layer height.

If I set the extrusion width manually to 0.40 mm, I then receive the error: "Flow::spacing() produced negative spacing. Did you set some extrusion width too small?"

These errors are unexpected, as my settings should theoretically be printable:
Extrusion width (0.40 mm) is greater than the layer height (0.30 mm).
Setting Default Extrusion Width to 100% should match the nozzle diameter (0.4 mm), which should work well with a layer height of 0.3 mm.

Steps Taken to Troubleshoot:
Manually Set Extrusion Width:
Tried setting extrusion width manually to 0.40 mm, resulting in the "Flow::spacing()" error.

Tested with Different Printer Profiles:
Tried slicing with different printer profiles (both default Prusa MK3 and custom Orion A150), but the issue persisted.

Checked Advanced Settings:

Tried Different Versions of PrusaSlicer:
This has been happening as early as 2.4 I believe.  Probably related to Arachne.

Expected Behavior:
When setting Extrusion Width to 100%, the slicer should correctly interpret this as the nozzle diameter (0.40 mm).
The slicer should not produce "extrusion_width=0.3 mm" or "Flow::spacing()" errors if extrusion width and layer height are both logically compatible.

Actual Behavior:
PrusaSlicer incorrectly calculates extrusion width as 0.3 mm (or produces "Flow::spacing()" errors when manually set to 0.4 mm).
Slicing is impossible with the current 0.30 mm layer height setting, even though all physical parameters should allow for it.

Screenshots:
[image]
[image]

Error Messages: 
When setting Default extrusion width manually to 0.40mm
[image]

When setting Default extrusion width to 100%
[image]

Steps to Reproduce:
Open PrusaSlicer (version 2.8.1).
Load Printer Profile with Nozzle Diameter set to 0.40 mm.
Set Layer Height to 0.30 mm.
Set Default Extrusion Width to 100% or manually set to 0.40 mm.
Try to Slice the Model (you will be presented with error either before slicing or when attempting)
Observe the extrusion width error or Flow::spacing() error.

Additional Information:
The issue persists across different printer profiles, both default and custom.
Nozzle size and extrusion width settings should not logically create a conflict given their relationship to the layer height.

### Project file & How to reproduce

[failed extrusion width.zip](https://github.com/user-attachments/files/17687402/failed.extrusion.width.zip)

**Issue Description:** I'm encountering a persistent issue in PrusaSlicer when slicing a model with the following settings:
- Nozzle Diameter: 0.40 mm
- Layer Height: 0.30 mm
- Default Extrusion Width: 100% (intended to match the nozzle diameter)

When I attempt to slice the file with these settings, I receive the following errors:

1. "ERROR: extrusion_width=0.3 mm is too low to be printable at a layer height 0.3 mm" – This happens even though the extrusion width should automatically set to 0.4 mm, which is greater than the layer height.
2. If I set the extrusion width manually to 0.40 mm, I then receive the error: "Flow::spacing() produced negative spacing. Did you set some extrusion width too small?"

These errors are unexpected, as my settings should theoretically be printable:
- Extrusion width (0.40 mm) is greater than the layer height (0.30 mm).
- Setting Default Extrusion Width to 100% should match the nozzle diameter (0.4 mm), which should work well with a layer height of 0.3 mm.

**Steps Taken to Troubleshoot:**

**1. Manually Set Extrusion Width:**
- Tried setting extrusion width manually to 0.40 mm, resulting in the "Flow::spacing()" error when trying to slice.  Sometimes if you increase to 0.50mm the error will go away, or if you change the angle of the print or change the scaling of the print, but this is not desirable.

**2. Set Extrusion width to 100%:**
- this will result in the error Invalid Data: "ERROR: extrusion_width=0.3 mm is too low to be printable at a layer height 0.3 mm".  If you change the layer height to 0.2mm, it will produce the same error except "ERROR: extrusion_width=0.2 mm is too low to be printable at a layer height 0.2 mm".  This shows that the Arachne generator is trying to match the extrusion width to the layer height for some reason when Extrusion Width is set to 100%.  
 
**3. Tested with Different Printer Profiles:**
- Tried slicing with different printer profiles (both default Prusa MK3 and custom Orion A150), but the issue persisted.

**4. Perimeter Generator Change:**
- Switched from Arachne to Classic perimeter generator.  This resolved the issue, however, I need Arachne!

**5. Tried Different Versions of PrusaSlicer:**
- This issue has occurred since 2.4 as far as I remember.  I have still yet to find a solution.
 
**Expected Behavior:**
When setting Extrusion Width to 100%, the slicer should correctly interpret this as the nozzle diameter (0.40 mm).
The slicer should not produce "extrusion_width=0.3 mm" or "Flow::spacing()" errors if extrusion width and layer height are both logically compatible.

**Actual Behavior:**
PrusaSlicer incorrectly calculates extrusion width as 0.3 mm (or produces "Flow::spacing()" errors when manually set to 0.4 mm).
Slicing is impossible with the current 0.30 mm layer height setting, even though all physical parameters should allow for it.

Screenshots:
[image]
[image]

Error Messages:
When default extrusion width is set to 100% 
[image]

When default extrusion width is set to 0.40mm
[image]

**Steps to Reproduce:**
- Open PrusaSlicer (version 2.8.1).
- Load Printer Profile with Nozzle Diameter set to 0.40 mm.
- Set Layer Height to 0.30 mm.
- Set Default Extrusion Width to 100% or manually set to 0.40 mm.
- Slice the Model.
- Observe the extrusion width error or Flow::spacing() error.

**Additional Information:**
The issue persists across different printer profiles, both default and custom.
Nozzle size and extrusion width settings should not logically create a conflict given their relationship to the layer height.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

Orion A150, Prusa MK3

u89djt, 2024-11-09:
(Fellow user following the hub)
If you increase Minimum Perimeter Width to at least 28% in the project file you posted, it works fine. I haven't tried to work out why yet.
[working extrusion width.zip](https://github.com/user-attachments/files/17688564/working.extrusion.width.zip)
[image]
[image]

adamrumjahn, 2024-11-09:
Yes! that resolved it! thank you

u89djt, 2024-11-09:
Very glad to hear it.
I suppose the question remains as to whether the slicer should be expected to prompt with a warning. 
The difficulty might be that multiple settings contribute to the feasibility, so the warning would look like an inequality of the form "ensure that a * b / c > 0.01, or automatically increase this value to d" or similar.

adamrumjahn, 2024-11-12:
Unrelated question: after slicing this file, I checked the "Actual Print Speed" and noticed something odd.  Even though I set the external perimeter acceleration to 25mm/s2, it seems to do the edges at highly inconsistent "Actual Speeds".  I'm trying to figure out what is causing this.  I don't think its just a visualization error, but this seems to also be reflecting reality because the difference in "Actual Speed" does indeed make a difference on the quality and I'm trying to get the "Actual Speed" to be as consistent as possible between layers.  Do you know what could be causing these differences in "Actual Speed"?  Is it related to the line segments? Any ideas on how I can remedy this?
[image]

Attached is the project file for reference.
[241108-ISOMED cranial implant equipment test (no holes)-13.zip](https://github.com/user-attachments/files/17713609/241108-ISOMED.cranial.implant.equipment.test.no.holes.-13.zip)

u89djt, 2024-11-12:
Hi there, this isn't the forum for teaching how to use the slicer per se, so just your starting points:
Zoom in on the slice preview and note the short ticks of filament that need to be started at precise locations. Ask yourself how precision is achieved and why the filament has to be laid that way; necessitated by features of the model.
Then consider what you need to know. Type "slow" into the search box.

adamrumjahn, 2024-11-12:
Hi Dave, thanks for your response.

I’ve taken a deeper look into this with the help of a few colleagues, and we’ve found that the issue seems to be related to retraction settings. Specifically, if I remove retraction or set "Minimum Travel After Retraction" to 0 mm, the "Actual Speed" becomes much more consistent. However, I'm still puzzled as to why retraction should impact the starting speed of an XY movement like this. When retraction is off, the speed starts higher; with it on, it starts lower, at least according to the visualization.

What I'm struggling with now is understanding if this difference is genuinely reflected in reality or if it's a quirk of the slicer's visualization. If it's a visualization inaccuracy, it might warrant being reported as an issue. If it's accurate, I'd love to understand the rationale behind it. In either case, I’d be happy to contribute to clarifying this behavior in the documentation, so others can understand it better.

Could you advise on where I might have a more in-depth dialogue with someone knowledgeable about these aspects of the slicer? I’m eager to learn and would also be willing to help improve the documentation based on what I find.

Thanks again for your help.

u89djt, 2024-11-12:
You've asked for my advice for something that isn't a slicer problem. This isn't the place for that.
One final comment from me: the slicer is doing everything it should, and the visualization is showing everything as it should.
https://forum.prusa3d.com/
https://help.prusa3d.com/article/customer-support_2287

adamrumjahn, 2024-11-12:
Hi Dave,

We looked into this a bit deeper and we believe there may be a visualization issue within the slicer related to how retractions affect the "Actual Speed" estimation. Specifically, it appears that the retraction interrupts the estimated speed calculation, leading to inconsistent speeds being visualized.

It's not really a functional issue with the slicer, just a visualization error as the correct accelerations are indeed applied, however, I don't believe that the visualizations accurately depict the actual speed.

Since this does indeed seem to be related to the software (I'm not here for a discussion on how to use Prusa slicer), I think it might be beneficial to bring this up as a new issue for further examination by the development team.

Jan-Soustruznik, 2025-01-02:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13429

closed 2025-01-02 by Jan-Soustruznik