prusa3d/PrusaSlicer #13576: Nozzle path causes floating (not attached) filament

FrHePo, 2024-11-09:
### Description of the bug

The sliced nozzle path leads to floating filament at Z=5mm.
[image]

### Project file & How to reproduce

[My X-carriage linear for Orbiter 2.0+RM PET-CF 0.6n.zip](https://github.com/user-attachments/files/17689637/My.X-carriage.linear.for.Orbiter.2.0%2BRM.PET-CF.0.6n.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer Version:   2.8.1+win64 Build:     PrusaSlicer-2.8.1+win64-202409181359  

### Operating system

Operating System:    Windows System Architecture: 64 bit Windows Version:     Windows 10 (build 19045), 64-bit edition

### Printer model

Prusa i3 MK3

neophyl, 2024-11-10:
I think this is the same issue as raised here https://github.com/prusa3d/PrusaSlicer/issues/13144
If you change the slicing to Classic rather than Arachne then it changes the perimeter order and does things in a more 'sensible' order.  

However that model geometry could be designed better to force any slicer to print better, it is still an unsupported hole in the middle of a flat surface.  The same modelling techniques used for counterbore holes would work here.

FrHePo, 2024-11-25:
Yes, that's right. Thanks, I successfully optimized the hole setting but still I feel that Arachne should be corrected to do better.

closed 2026-09-29 by prusaslicer-bot[bot]