prusa3d/PrusaSlicer #13577: Gcode for ender 3 having /n instead of new lines in the gcode.

Dtwpurple, 2024-11-10:
### Description of the bug

the gcode being produced by prusa is now messed up and not functioning. so the start temps and everything are all not functioning becuase they have /n in front of them.  theres random /n in the code instead of new lines in the gcode. 

### Project file & How to reproduce

[keytestx10.zip](https://github.com/user-attachments/files/17690239/keytestx10.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

newest

### Operating system

windows 11 newest at the time of upload

### Printer model

ender 3 pro

Dtwpurple, 2024-11-10:
honestly its been such a pain. im just switiching to cura. hope this gets fixed one day tho. (cura actually produces files that dont have this anomaly...)

u89djt, 2024-11-10:
@Dtwpurple if you can stick with this, it'll help a lot of users like me. I assume you're using the latest version of the slicer, and it didn't have this problem previously? You didn't say which you're using. If you can post a project file, they can make certain they're dealing with your specific problem. You might be surprised how many small settings changes have important impacts around the hub. I think in this case it's obviously a text translation slip, but there's not enough information to be sure your specific situation is definitely going to be addressed in this thread per se.

u89djt, 2024-11-10:
I'm a doofus - it says 2.8.1 in the gcode. Post the project tho.

rtyr, 2024-11-10:
I don't see anything wrong with the file you uploaded. If you see `\n` instead of lines, you are likely viewing the g-code in the text editor which does not interpret line breaks.

As for the issues with the temps, I believe you have your start g-code set incorrectly:
1) You are using hardcoded value for hotend temperature (always 200C) instead of using temperature from filament settings.
2) M109 commands are completely missing, so printer does not wait for temperatures and start printing prematurely...

If you are not sure how to set up start g-code correctly, you can enable and try bundled Ender-3 profiles.

> so the start temps and everything are all not functioning

That is likely because you start gcode is incorrect. I can see it sets temperature of the hotend always to 200

Dtwpurple, 2024-11-10:
Nah that’s not how it works. /n shouldn’t be there. The printer is reading the code just like we do from the txt file. If there’s a /n the printer does not use that command because there is no/n command for marlin. If you look at other code produced from other slicers they don’t do this. But Prussia does cause it has some parser issue or something. Some how the code isn’t being produced right. It’s an inherent flaw that got produced by a prusa update. My printer accepts temps but they don’t change when using prusa slicer…. Because the gcode compiler is messed up. LmaoBut cura works just fine? Ya like I said I’m over this good luck

rtyr, 2024-11-10:
> Nah that’s not how it works. /n shouldn’t be there. 

It is not there, in the gcode you provided.

rtyr, 2024-11-10:
> but they don’t change when using prusa slicer

Yes, like I explained, your start g-code is incorrect.

Your start g-code looks like this 
```
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S200 ; set temporary nozzle temp to prevent oozing during homing
M140 S{first_layer_bed_temperature[0]} ; set final bed temp
G4 S30 ; allow partial nozzle warmup
G28 ; home all axis
G29
G1 Z50 F240
G1 X2 Y10 F3000
G1 Z0.28 F240
G92 E0
G1 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0
```
While M104 command should like this
`M104 S{first_layer_temperature[0]}`

>  Ya like I said I’m over this good luck

I guess we can close this issue, since there is no issue with the provided g-code and the main issue is in your printer profile.

u89djt, 2024-11-11:
Fleshing out the user perspective that might motivate future submissions to follow the guidelines.
The object being sliced is Creative Commons Attribution: https://www.thingiverse.com/thing:4883022
If the original poster had provided the project file, they would have got something like this:
"You've entered bespoke settings for your printer (presumably saved as "Creality Ender-3 internetsettings", which is different to the 3 pro mentioned by the user. I don't know how different those profiles might be), but there's a profile for it (I've gone with the pro) distributed with the slicer. You install it like this:
Open up the wizard
[image]
Show third party FFF printer files
[image]
Activate the 3 pro profile
[image]
There's no Amazon basics filament profile included for the 3 pro so I selected Generic PLA and had a go at matching the object and slicing parameters.
[image]
It looks like the object is identical, and the support is largely identical, so I assume it will do. 
[ender3probundledJackSallyextrusionmatch.zip](https://github.com/user-attachments/files/17702759/ender3probundledJackSallyextrusionmatch.zip)
[image]
That profile is bundled with the slicer, and isn't marked as Alpha or Beta, so I would expect it to work for you.
"
This would make it possible for a user unfamiliar with gcode and Marlin to go ahead and print on the 3 pro.
An expert like Roman might still have diagnosed the user's information deficit after the solution was provided if it seemed worthwhile, but the user would have had their problem solved more efficiently, and without distracting an expert from their work.
Specifically for my activities: I probably didn't help matters by speculating about text translation. That's a lesson for me. It was a confusion/misinterpretation of text by the user, not the slicer, and I need to think more about what might be inferred if I'm going to assert something where I haven't engaged with the relevant literature.
In this case, there is ready reference for Marlin firmware describing the use of the "\n" string:
[image]