prusa3d/PrusaSlicer #13581: Dice Dagger Failure

nimartet, 2024-11-12:
### Description of the bug

I was slicing the [Dice Dagger Model](https://www.printables.com/model/877226-print-in-place-collapsing-dice-dagger) in Prusa Slicer 2.8.1 with the following settings: 

- Printer: Anycubic Photon Mono X 6K
- SLA print settings: 0.05 Normal
- SLA material: Anycubic Plant-based UV Resin Grey 23
- Supports: Everywhere
- Pad: Around object

The status bar hangs for a while at 48% and indicates "Generating support tree."  After a while, we get to about 90% and a message box appears:  

[image]

At that point, I have 42% memory utilization.  So I have more than half of my RAM available to be used.

(Note: I am adding tree supports using this method and then exporting the stl with supports in order to import, slice, and print on my MK3S+)

Device and OS info:

Device name	MSI
Processor	Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
Installed RAM	16.0 GB (15.8 GB usable)
Device ID	9EF58693-E59B-4F8D-B9FF-CD999F2AAB74
Product ID	00325-81655-65384-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen support

Edition	Windows 11 Home
Version	23H2
Installed on	‎9/‎22/‎2022
OS build	22631.4391
Experience	Windows Feature Experience Pack 1000.22700.1047.0

### Project file & How to reproduce

To repeat, download the  [Dice Dagger Model](https://www.printables.com/model/877226-print-in-place-collapsing-dice-dagger) and use with the project settings above and slice.

[DICE DAGGER PNP Report.zip](https://github.com/user-attachments/files/17709296/DICE.DAGGER.PNP.Report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+win64

### Operating system

Windows 11 Home

### Printer model

Anycubic Photon Mono X 6K is the selection

nimartet, 2024-11-12:
> (From a fellow user who appreciates meeting the idea of saving a tree
> supported object to stl for FFF printing)
> To follow the submission instructions, save the project just before you
> carry out the action that leads to the crash to 3mf, zip it, and post that.

Hello Dave, I'm sorry. I thought I attached the zip file. Was there
something wrong with it?

u89djt, 2024-11-12:
I'm a buffoon - I was overtaken by enthusiasm to try your approach and glitched :). It's absolutely fine, sorry!

u89djt, 2024-11-12:
On my machine, it succeeds:
[image]
I'm running 32 gigs of RAM and a beefy GPU setup, so maybe that saved me.

u89djt, 2024-11-12:
Main RAM usage caps out at about 4.5 gig on your project.
Are you running 11 gigs of other stuff? Lemme check GPU occupation...

u89djt, 2024-11-12:
Less that 3 gig at max and honestly I don't know if that's relevant.
So it works ok on:
Version: 2.8.1+win64
Build: PrusaSlicer-2.8.1+win64-202409181359
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 33,435MB
PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 33,435MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 565.90)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 4090/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 335 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

nimartet, 2024-11-12:
> ( @nimartet <https://github.com/nimartet> I assume you know this, but
> assumptions scupper ships: you've tried the native organic supports for
> FFF? )

Interesting!

If I remember correctly linear supports generated fine, but my projects may
be blurring in my head lol.

It sounds like I need more RAM? When I start the project I am at 43%
utilization. I've only got 16 gigs. So that leaves me with only a little
over 9 gigs left. If more is needed then that then this won't work for me
lol. Although when monitoring the memory I don't see it go that high
strangely.

When I get off of work I will try the linear supports again just to verify.
I will also try to see if I can increase my virtual memory to see if that
helps.

I wonder if there is a caching technique or some other programming paradigm
that can be used to improve the application to limit its RAM usage?

But at any rate, since it's a mem issue, I guess we can close this ticket.
Thanks for trying it out!

u89djt, 2024-11-12:
No worries! Maybe leave it open for a couple of days to give folk a chance to see it and recognize something. There have been processor issues here and there - the internet knows infinitely more than I do :)

u89djt, 2024-11-12:
Here's what default organic support settings generate:
[image]
And obviously they can be altered pretty extensively.
I think that model will print without any supports apart from maybe a brim.

nimartet, 2024-11-12:
Wait, you can select the support type? I guess I didn't know you could do
that (or I forgot...both are likely). I had read an article about how to
get tree supports using a different printer option and exporting/importing
the STL with supports and front using your FFF.

nimartet, 2024-11-12:
And yes the model prints without supports. I got it done in 19.5 hours but some of the bridging of compartments in the hidden blade was too expansive and it sagged.

u89djt, 2024-11-12:
Yeah, new stuff pops up over time :) Time to check in on the knowledge base: https://help.prusa3d.com/tag/mk3s/prusaslicer_204
Have fun!

u89djt, 2024-11-12:
Oh hey, have a look at infill options, too. Have you seen "lightning?"

nimartet, 2024-11-12:
I increased my virtual memory:
[image: virt-men.png]
I still failed to generate the tree supports.  It crashes after about 86%
complete.

Below is a screenshot of my mem usage at the time of the crash (when the
message box appears):

[image: image.png]

But I think that is wrong because my system completely wigs out (I could
not open a browser for example until i closed the prusa slicer).

I easily and quickly generated the grid, organi, and snug supports:

[image: image.png]
[image: image.png]
[image: image.png]

Oh well.  I guess organic supports it is!

u89djt, 2024-11-12:
Looks like you posted by replying to email? Screenshots don't come through that way, you need to use the web client.

nimartet, 2024-11-12:
I tried updating the comment above but something didn't go right. But anyway you can still click on the links to see the images.

u89djt, 2024-11-12:
Yep, those images are reachable. 
I've no idea at this point what's grinding is gears.

TerPuc, 2025-05-27:
Hi @nimartet, thank you for reporting! We have created internal issue with number SPE 2771 and we'll look at it.

nimartet, 2026-09-18:
I was told it was a memory issue and I have not had a need to use this technique so it has not been tried again.

closed 2026-09-18 by nimartet