prusa3d/PrusaSlicer #13584: OOM calculating overhanging parameters

huddlesj, 2024-11-12:
### Description of the bug

Trying to slice coasters from https://www.printables.com/model/671674-snowflake-coasters-hueforge/files

[image]

My work station with 64 Gig of ram gets an OOM error. 

[image]

### Project file & How to reproduce

[prusa-snowflake-coasters-set2.zip](https://github.com/user-attachments/files/17723821/prusa-snowflake-coasters-set2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

Prusa XL 5t

u89djt, 2024-11-13:
(data point from another user)
I'm only running 32 gig of RAM and it succeeds for me, topping out at just over 2 gig of RAM usage.
[image]
More info about your setup is needed. What's your processor? Processor issues have cropped up in the past; https://github.com/prusa3d/PrusaSlicer/issues/13483#issuecomment-2411517036 , for example.

u89djt, 2024-11-13:
This is the warning group:
[image]
Presumably because of these.
[image]
That situation - where some layers in a monotonically reducing cone fail to result in extrusions under others that do extrude - is new to me.

huddlesj, 2024-11-14:
Desktop is a Intel NUC Extreme 13. Intel Core i9-13900k processor, 64 gig of ram and a 3060 video card.

huddlesj, 2024-11-18:
I think i found a work around. If I create three instances of each coaster it slices just fine. 
[image]

closed 2026-09-29 by prusaslicer-bot[bot]