prusa3d/PrusaSlicer #13590: Slice introduces overhang perimeter in a place it shouldn't

christophe-calmejane, 2024-11-13:
### Description of the bug

I have an object I'm slicing and it generated some overhang perimeter instead of solid layers.

This is the model:
[image]

And this is how it looks when sliced:
[image]

### Project file & How to reproduce

[Bug_Overhang.zip](https://github.com/user-attachments/files/17739244/Bug_Overhang.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1-win64

### Operating system

Windows 10

### Printer model

XL-2tools

u89djt, 2024-11-13:
Yep, cut it under the bridge and you can see
[image]

u89djt, 2024-11-13:
Less work: paint and clip view
[image]

christophe-calmejane, 2024-11-15:
Ok then, confirmed as a bug in the model, I notified the creator which confirms (and already fixed it). Sorry for the incorrect report, since the other very similar model I had was fine, I thought it was a bug in the slicer.