prusa3d/PrusaSlicer #13591: Negative X coordinates showing up in g-code.

dtmille2, 2024-11-14:
### Description of the bug

Sliced a model and it generated negative x coordinates during a move at the end of one of the layers.  Occurs on lines 17217 through 17232.  Causes the print to crash on my Creality printer.

### Project file & How to reproduce

[Project File.zip](https://github.com/user-attachments/files/17742180/Project.File.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.1 +win64

### Operating system

Windows 10     Version	10.0.19045 Build 19045

### Printer model

Creality CR-10 SE

dtmille2, 2024-11-14:
[image]

u89djt, 2024-11-14:
They're on it: https://github.com/prusa3d/PrusaSlicer/issues/13591 
[image]
There are some pretty wild moves in your project.
[image]

u89djt, 2024-11-14:
Just confirming here that if you un-tick "avoid crossing perimeters" those off-bed moves don't appear.
[image]

dtmille2, 2024-11-14:
Hi Dave, I will check that and let you know thanks!

image.png (view on web)<https://github.com/user-attachments/assets/54abc0dd-3f51-45c2-a1b0-5f5de500a86f>

u89djt, 2024-11-14:
Just post any new info you find, don't bother if it seems like you're covered (or if a dev asks :) ). You'll want to do screenshots and such if you - those need to come via the web interface.

Jan-Soustruznik, 2024-11-22:
Hello, @dtmille2. This issue seems similar to #13478 and should be resolved in the latest alpha release (2.9.1-alpha1): [PrusaSlicer 2.9.0-alpha1 Release.](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1)

Please check if the issue persists with this version. For now, I will close this issue as a duplicate. However, if you encounter the same problem with the latest release on any 3MF file, feel free to create a new issue and share your project file.

Thank you for your report, and have a great day!

closed 2024-11-22 by Jan-Soustruznik