prusa3d/PrusaSlicer #13592: Slicer crashes when slicing certain models

Tossrock, 2024-11-15:
### Description of the bug

When attempting to slice a model (see attached project), the program crashes with no messages.

The program should not crash.

### Project file & How to reproduce

Steps to reproduce:

1. Open the project with PrusaSlicer-2.8.1 win64.
2. Click "Slice now"
3. The program crashes.

[Prusa_Project_pre_crash.zip](https://github.com/user-attachments/files/17761351/Prusa_Project_pre_crash.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

Prusa Mk4

u89djt, 2024-11-15:
(fellow user following the hub)
It slices ok for me windows 11, ps 2.8.1, ryzen
[image]
Is there an error message when it crashes?
What's your processor? There have been CPU issues here and there, e.g. https://github.com/prusa3d/PrusaSlicer/issues/13483#issuecomment-2411517036

Tossrock, 2024-11-19:
> (fellow user following the hub) It slices ok for me windows 11, ps 2.8.1, ryzen ![image](https://private-user-images.githubusercontent.com/75209005/386634328-feae1aa2-8926-4dcc-8fab-6469a23a85a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5ODE0MzQsIm5iZiI6MTczMTk4MTEzNCwicGF0aCI6Ii83NTIwOTAwNS8zODY2MzQzMjgtZmVhZTFhYTItODkyNi00ZGNjLThmYWItNjQ2OWEyM2E4NWE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDAxNTIxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlNDVlYTk5MmM2OWQzZDNiYjlhNDQzNDk5ZjdjMWYyNDY3YWI3Y2E0MzI5YTc3MjA2ZjdkNWUwNzM4NDM2NjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.LRsaOPcCLll40QVuGRUsyQ8gMXFq6l7nNcEXO3_1NkA) Is there an error message when it crashes? What's your processor? There have been CPU issues here and there, e.g. [#13483 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13483#issuecomment-2411517036)

Searching around, I think you're right - I have an Intel 14900k on a Z790 mobo, and I was able to fix it by pinning the core associativity away from the latter half of the cores. Thanks for nothing, Intel!

closed 2024-11-19 by Tossrock