prusa3d/PrusaSlicer #13596: "Split" moves the parts around

salamanders, 2024-11-16:
I'm trying to split a CrystalWing dragon to put a new tail on it.  When I Split the model, the split works - but then all the parts are shifted slightly.

Can the "split" not move parts around?

neophyl, 2024-11-16:
Are you using Split to Object or Split to Parts ?  Split to Objects will place all the now separated Objects to the build plate.  As ALL Objects must touch the plater.  Split to Parts on the other hand shouldn't move them relative to each other when split.  And a Part can be placed wherever as long as Some portion of the Object still touches the plater.

salamanders, 2024-11-17:
I'm using "Split" in the latest Prusa Slicer - I don't see an option to choose between "Split to Object" vs "Split to parts"

OH INTERESTING.  Depending on how I load the file, it doesn't give me the option!  If I load the MMU as separate objects to start, it gives me the option to split to parts!  Very cool!

closed 2024-11-17 by salamanders