prusa3d/PrusaSlicer #13599: Multiple parts wont slice, crash 100%

Steeeeev, 2024-11-17:
### Description of the bug

I have tried to download some popular models from printables and slice them all at once. The progress crashes every time I attempt to slice the parts. This screenshot has 5 models in it, but I can reproduce the crash with just 2 of them pretty regularly. I have a .6 nozzle installed, but other than that the print profile settings are unchanged in any way. What am I doing wrong?

### Project file & How to reproduce

[Error Report file.zip](https://github.com/user-attachments/files/17791205/Error.Report.file.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Win 11

### Printer model

MK3s+

u89djt, 2024-11-17:
That project slices fine for me (another user). What does the crash look like? Is there an error message?
[image]

Steeeeev, 2024-11-17:
> That project slices fine for me (another user). What does the crash look like? Is there an error message?
> [image]
> 

Says illegal settings or something. It dissappear too fast. I flashed the BIOS on my pc mb and it sliced so maybe my hardware was acting up? Anyway. Thanks for confirming it was me.

closed 2024-11-17 by Steeeeev

Steeeeev, 2024-11-17:
> No worries. For future reference: if you get a repeatable error that's too fast to catch, you could video the screen with your phone. It might also appear in the "event viewer". If you can say which processor you have, and which BIOS version old/new, someone with the same problem (you're probably not alone) might find this and their answer. Cheers!

Intel 14900k cpu, MSI Z790 MEG ACE MB old BIOS 7D86v19  new BIOS 7D86v1D. Hope it helps

lukasmatena, 2024-11-19:
This is very likely the same as #11845. You may find a remedy if you go through that discussion.