prusa3d/PrusaSlicer #13607: Skip This Update-button on "App Update Available"-popup

MNeMoNiCuZ, 2024-11-20:
**Is your feature request related to a problem? Please describe.**
The "App Update available" pop-uphas 3 functionalities currently:
1. Cancel (closes the window, it will pop up next time you start Slicer)
2. Next (download/install update)
3. Don't notify about new releases any more

**Describe the solution you'd like**
I would like to add a 4th behavior:
"Skip this update"

**Describe how it would work**
When "Skip this update" is pressed (it's a button), the pop-up goes away until the next update is available.

**Describe alternatives you've considered**
It would be great to see a one-sentence quick summary of the patch notes. So that I can see what's new in the latest iteration.
This way I can make an informed decision right in the UI of the program.

**Additional context**
This type of behavior exists in other programs.
An example is "4k Video Downloader". They show a changelog, and lets me skip this update, or install now, or just close.

MNeMoNiCuZ, 2026-09-04:
Impossible to test until the new version has auto-updates.
I would bet this is not yet fixed though.

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion https://github.com/prusa3d/PrusaSlicer/discussions/16141.

closed 2026-10-21 by Jan-Soustruznik