prusa3d/PrusaSlicer #13610: [Alpha Release] Error Opening OrcaSlicer Project in PrusaSlicer: Model Issue

Zogar89, 2024-11-20:
### Description of the bug

Open OrcaSlicer Project shows
Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 21
[image]

### Project file & How to reproduce

[E Orca.zip](https://github.com/user-attachments/files/17837152/E.Orca.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Alpha 2.9

### Operating system

Windows 10

### Printer model

Artillery x2

Zogar89, 2024-11-20:
That file shows...

```
<?xml version="1.0" encoding="UTF-8"?>
<model unit="millimeter" xml:lang="en-US" xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02" xmlns:BambuStudio="http://schemas.bambulab.com/package/2021" xmlns:p="http://schemas.microsoft.com/3dmanufacturing/production/2015/06" requiredextensions="p">
 <metadata name="Application">BambuStudio-2.2.0</metadata>
 <metadata name="BambuStudio:3mfVersion">1</metadata>
 <metadata name="Copyright"></metadata>
 <metadata name="CreationDate">2024-11-20</metadata>
 <metadata name="Description"></metadata>
 <metadata name="Designer"></metadata>
 <metadata name="DesignerCover"></metadata>
 <metadata name="DesignerUserId"></metadata>
 <metadata name="License"></metadata>
 <metadata name="ModificationDate">2024-11-20</metadata>
 <metadata name="Origin"></metadata>
 <metadata name="Thumbnail_Middle">/Metadata/plate_1.png</metadata>
 <metadata name="Thumbnail_Small">/Metadata/plate_1_small.png</metadata>
 <metadata name="Title"></metadata>
 <resources>
  <object id="3" p:UUID="00000001-71cb-4c03-9d28-80fed5dfa1dc" type="model">
   <components>
    <component p:path="/3D/Objects/E.stl_1.model" objectid="1" p:UUID="00010000-b206-40ff-9872-83e8017abed1" transform="1 0 0 0 1 0 0 0 1 0 0 0"/>
    <component p:path="/3D/Objects/E.stl_1.model" objectid="2" p:UUID="00010001-b206-40ff-9872-83e8017abed1" transform="1.33333333 0 0 0 1.33333333 0 0 0 0.266666667 0.768953869 14.050871 -1.36581429"/>
    <component p:path="/3D/Objects/E.stl_1.model" objectid="2" p:UUID="00010002-b206-40ff-9872-83e8017abed1" transform="0.833333333 0 0 0 0.833333333 0 0 0 0.2 -1.61414123 -26.3730371 -27.0270428"/>
    <component p:path="/3D/Objects/E.stl_1.model" objectid="2" p:UUID="00010003-b206-40ff-9872-83e8017abed1" transform="1.83333333 0 0 0 1.83333333 0 0 0 0.2 0.768953869 14.050871 5.90557923"/>
    <component p:path="/3D/Objects/E.stl_1.model" objectid="2" p:UUID="00010004-b206-40ff-9872-83e8017abed1" transform="1.16666667 0 0 0 1.16666667 0 0 0 0.2 -1.61414123 -26.3730371 -21.287267"/>
   </components>
  </object>
 </resources>
 <build p:UUID="2c7c17d8-22b5-4d84-8835-1976022ea369">
  <item objectid="3" p:UUID="00000003-b1ec-4553-aec9-835e5b724bb4" transform="1 0 0 0 1 0 0 0 1 113.426491 148.140347 48.6170006" printable="1"/>
 </build>
</model>
```

Jony01, 2024-11-25:
Long story short: 3MF goes aginst specification.

In file `E Orca.3mf\3D\Objects\E.stl_1.model` is object with `id="2"` defined with `type="other"`. This type by 3mf core specification can't be addressed from `build/item`.

> Note: items MUST NOT reference objects of type "other", either directly or recursively.

[link  to 3mf core specification](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#3431-item-element)

closed 2024-11-25 by Jony01

Zogar89, 2024-11-25:
I will open this issue in Orca git to tell that.