prusa3d/PrusaSlicer #13614: 2.9.0-alpha1 Fuzzy Skin and Seam Painting do not work together

ex-nerd, 2024-11-21:
### Description of the bug

When using the new paint-on fuzzy skin feature, it seems that seam configuration is ignored and defaults to … I'm not sure what. Placement of the seam on any layer with fuzzy skin is not where you would expect (confirmed with `aligned`, `rear`, and painted seams).

### Project file & How to reproduce

Sample project: 
[fuzzy-skin-seam-painting-bug.3mf.zip](https://github.com/user-attachments/files/17839446/fuzzy-skin-seam-painting-bug.3mf.zip)

1. Add a random object
2. Paint a random area with fuzzy skin
   - [image]
3. Paint a seam in a specific spot
   - This isn't actually necessary but gives a good example. You can also see this effect with normal seam settings like `aligned` or `rear`.
   - [image]
5. Slice and turn on seam visibility
6. Notice that the seam is in an unexpected location for any layer that also includes fuzzy skin
   - [image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0-alpha1

### Operating system

MacOS 15.1.1

### Printer model

Prusa XL 5 tool

Jan-Soustruznik, 2024-11-21:
Hello, @ex-nerd  , thank you for report. I created internal issue no. SPE-2576. We will try to fix in some next release. Have a nice day

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo