prusa3d/PrusaSlicer #13616: Small Items on a Raft have overlapping Raft which shows an error

BrienKing, 2024-11-21:
### Description of the bug

Small Items on a Raft have overlapping Raft which shows an error.

[image]

### Project file & How to reproduce

Slice this 3MF file.

[100x Prusa MK3S+ .4N .3L ABS Any Coin Pegs.zip](https://github.com/user-attachments/files/17842177/100x.Prusa.MK3S%2B.4N.3L.ABS.Any.Coin.Pegs.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

MK3S+ but it has the same issue on my XL

u89djt, 2024-11-21:
(fellow user following the hub)
It's not because they're small, it's because they're positioned in a way that means the rafts overlap. This happens with any aspect of support when distinct objects are close to each other. You either have to move them further apart as the warning suggests, or merge them into a single object so the raft is generated for everything together.
Set them all as separate objects
[image]
Consider optimizing the arrangement
[image]
Then select them all and right click->merge. 
[image]
And the slice works
[image]

BrienKing, 2024-11-21:
Thanks for the work-around.   It would be nice that if it could detect the overlap and automatically merge the rafts.

BrienKing, 2024-11-21:
One caveat to this, each object has to be separate object in order to do the merge.  Setting the Instance count doesn't work.  What I did was, I selected all of the instances, right clicked on them, then selected "Set as Separated object".  I was then able to select all of those and select the "Merge".    This then generated the Raft the way I needed it.

u89djt, 2024-11-21:
Yep, first sreenshot shows the right click menu.

closed 2026-09-29 by prusaslicer-bot[bot]