prusa3d/PrusaSlicer #13617: 2.9.0 alpha multiple beds slicing and naming

darraghbr, 2024-11-21:
The multiple beds functionality in 2.9.0 looks very useful, I understand that slicing multiple beds in one step is upcoming but I thought I would start a discussion regardng the workflow of slicing and exporting mutlple beds. 

It would be really helpful if all the beds could be exported in a single step as well, the naming convention could be as simple as ProjectName_BedNumber.gcode however I think some clever naming could be implemented. 

Because many projects need different coloured parts it would be great if we could name the individual beds. So for instance if you had 3 beds worth of prints you could name them Red1, Red2, Black1 etc and then the 3 gcode files would be ProjectName_Red1.gcode, ProjectName_Red2.gcode etc. It could also be used to keep track of the number of times that bed needs to be printed for that project, say with 1off, 2off etc. Naming them would also help with MMU users as they could list the colours that needed to be loaded. 

This could open additional functionality like assigning a specific material profile to each bed, so if you had a bed of PETG parts and a bed of PLA parts that could be sliced and exported in a single workflow.

prowlmedia, 2024-11-25:
I'd like to see the output file name format in settings as a custom field per plate in the main interface 

[image]

rcmaniac25, 2024-11-25:
I too would like to throw into the pot for requesting the ability to name beds.

rcmaniac25, 2026-09-13:
Still relevant, especially now.