prusa3d/PrusaSlicer #13618: segfault in fresh installation as user (--user) prusaslicer fatpack 2.9.0-alpha

sl1pkn07, 2024-11-21:
### Description of the bug

after fresh install prusaslicer via fatpack as user

```
flatpak remote-delete flathub
flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo --user
flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo --user
flatpak --user install flathub-beta com.prusa3d.PrusaSlicer
```
and run it

```
flatpak run com.prusa3d.PrusaSlicer//beta
```

the program not open. instead, print this message in console

```
--------------------------------------------------------------------------
Message: 16:33:16: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
```

then run coredumpctlr ad see
```
└───╼  coredumpctl 
TIME                            PID  UID  GID SIG     COREFILE EXE                     SIZE
Thu 2024-11-21 16:29:39 CET 3052233 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
Thu 2024-11-21 16:29:39 CET 3052246 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
Thu 2024-11-21 16:33:16 CET 3053291 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
Thu 2024-11-21 16:33:17 CET 3053318 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
```

run  `coredumpctl info 3053318` (the last one run for example) and:
```
└───╼  coredumpctl info 3053318
           PID: 3053318 (prusa-slicer)
           UID: 1000 (sl1pkn07)
           GID: 1000 (sl1pkn07)
        Signal: 11 (SEGV)
     Timestamp: Thu 2024-11-21 16:33:16 CET (14min ago)
  Command Line: /app/bin/prusa-slicer
    Executable: /app/bin/prusa-slicer
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-flatpak-com.prusa3d.PrusaSlicer-3053274.scope
          Unit: user@1000.service
     User Unit: app-flatpak-com.prusa3d.PrusaSlicer-3053274.scope
         Slice: user-1000.slice
     Owner UID: 1000 (sl1pkn07)
       Boot ID: c30bf56595c444a685971615ce652f68
    Machine ID: c20ee0c57658685bfedf50384b0e3ec0
      Hostname: sL1pKn07
       Storage: /var/lib/systemd/coredump/core.prusa-slicer.1000.c30bf56595c444a685971615ce652f68.3053318.1732203196000000.zst (present)
  Size on Disk: 3.6M
       Message: Process 3053318 (prusa-slicer) of user 1000 dumped core.
                
                Module /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0.6.12 without build-id.
                Module /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0.6.12
                Module /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0.16.5 without build-id.
                Module /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0.16.5
                Stack trace of thread 9:
                #0  0x00007cc6202f2618 n/a (/usr/lib/x86_64-linux-gnu/GL/nvidia-565-57-01/extra/libnvidia-tls.so.565.57.01 + 0x2618)
                #1  0x00007cc633d04a75 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x34a75)
                #2  0x00007cc633d05022 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x35022)
                #3  0x00007cc633cfbdef n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x2bdef)
                #4  0x00007cc633cfa9a6 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x2a9a6)
                #5  0x00007cc633cfa568 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x2a568)
                #6  0x00005d945dff476f n/a (/app/bin/prusa-slicer + 0x2c9e76f)
                #7  0x00005d945dff7ce2 n/a (/app/bin/prusa-slicer + 0x2ca1ce2)
                ELF object binary architecture: AMD x86-64
```

### Project file & How to reproduce

fresh install fatpack via system package for the first time and:

```
flatpak remote-delete flathub
flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo --user
flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo --user
flatpak install flathub-beta com.prusa3d.PrusaSlicer --user 
```

run it

```
flatpak run com.prusa3d.PrusaSlicer//beta
```

and see the ouptut

```
--------------------------------------------------------------------------
Message: 16:33:16: Starting PrusaSlicer flatpak with entrypoint script
---

run `coredumpctl` and see the output

```
└───╼  coredumpctl 
TIME                            PID  UID  GID SIG     COREFILE EXE                     SIZE
Thu 2024-11-21 16:29:39 CET 3052233 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
Thu 2024-11-21 16:29:39 CET 3052246 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
Thu 2024-11-21 16:33:16 CET 3053291 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
Thu 2024-11-21 16:33:17 CET 3053318 1000 1000 SIGSEGV present  /app/bin/prusa-slicer   3.6M
 ```

and run `coredumpctl info $PID`

for see the info of the segfault

```
└───╼  coredumpctl info 3053318
           PID: 3053318 (prusa-slicer)
           UID: 1000 (sl1pkn07)
           GID: 1000 (sl1pkn07)
        Signal: 11 (SEGV)
     Timestamp: Thu 2024-11-21 16:33:16 CET (14min ago)
  Command Line: /app/bin/prusa-slicer
    Executable: /app/bin/prusa-slicer
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-flatpak-com.prusa3d.PrusaSlicer-3053274.scope
          Unit: user@1000.service
     User Unit: app-flatpak-com.prusa3d.PrusaSlicer-3053274.scope
         Slice: user-1000.slice
     Owner UID: 1000 (sl1pkn07)
       Boot ID: c30bf56595c444a685971615ce652f68
    Machine ID: c20ee0c57658685bfedf50384b0e3ec0
      Hostname: sL1pKn07
       Storage: /var/lib/systemd/coredump/core.prusa-slicer.1000.c30bf56595c444a685971615ce652f68.3053318.1732203196000000.zst (present)
  Size on Disk: 3.6M
       Message: Process 3053318 (prusa-slicer) of user 1000 dumped core.
                
                Module /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0.6.12 without build-id.
                Module /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0.6.12
                Module /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0.16.5 without build-id.
                Module /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0.16.5
                Stack trace of thread 9:
                #0  0x00007cc6202f2618 n/a (/usr/lib/x86_64-linux-gnu/GL/nvidia-565-57-01/extra/libnvidia-tls.so.565.57.01 + 0x2618)
                #1  0x00007cc633d04a75 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x34a75)
                #2  0x00007cc633d05022 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x35022)
                #3  0x00007cc633cfbdef n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x2bdef)
                #4  0x00007cc633cfa9a6 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x2a9a6)
                #5  0x00007cc633cfa568 n/a (/usr/lib/x86_64-linux-gnu/libc.so.6 + 0x2a568)
                #6  0x00005d945dff476f n/a (/app/bin/prusa-slicer + 0x2c9e76f)
                #7  0x00005d945dff7ce2 n/a (/app/bin/prusa-slicer + 0x2ca1ce2)
                ELF object binary architecture: AMD x86-64
```

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 alpha

### Operating system

Archlinux

### Printer model

none

SachCZ, 2024-11-22:
Hello @sl1pkn07, this is very hard for us to reproduce as it does not happen on any of the linux distributions we tested. It is especially hard, because flatpak should provide reproducible environment for the most part, so you probably stumbled upon something very specific.

Just from the limited segfault information, can you please make sure your graphics card drivers are properly installed? But honestly, that is grasping at straws.

Can you think of a way we could reproduce it? Since you are using Arch, I am afraid it might be hard to reproduce your setup.

sl1pkn07, 2024-11-22:
maybe making a fatpack with all symbols possible (sadly nvidia not provide their libaries with symbols) included can output a more specific output. but eats a tons and tons of system resources, included hdd space.

other option is start build again a "native" package myself like always i do in old times until star fail build because reasons not fixed in upstream.. idk if is fixed now. let me try this weekend.

greetings

SachCZ, 2024-11-22:
I am sorry to hear you struggle like this. Building might not work right now, but there is an open PR https://github.com/prusa3d/PrusaSlicer/pull/13609 that we plan to merge ASAP.

You are right about the symbols, we should do that. But we cannot obviously bundle symbols for everything that the flatpak is made of, so it might not help here.

SachCZ, 2024-11-22:
Thank you, this is more helpful. We will see what we can do.

SachCZ, 2024-11-22:
Just for reference, PrusaSlicer does not currently support wayland, but there is a PR: https://github.com/prusa3d/PrusaSlicer/pull/13307

bkerler, 2024-12-09:
I can confirm having the same issue, it seems to be related with the latest nvidia drivers 556 and opengl usage. I'm trying to compile prusaslicer right now to figure out the function causing the error.

bkerler, 2024-12-09:
Interesting. If I run the program normally, I do get a segfault. However if I run it with gdb or strace, it works without crashing.

bkerler, 2024-12-09:
using "__GLX_VENDOR_LIBRARY_NAME=mesa ./prusa-slicer" no segfault occurs. the segfault seems to be caused by nvidia-tls linkage. Kernel log says: `[23713.140495] traps: prusa-slicer[1965999] general protection fault ip:7f9005e50590 sp:7ffc4db40188 error:0 in libnvidia-tls.so.565.57.01[2590,7f9005e4f000+2000]`

bkerler, 2024-12-09:
Clearly not the way to do it, but after i removed the `/usr/lib/x86_64-linux-gnu/libnvidia-tls.so.565.57.01` and ran "sudo ldconfig" afterwards, everything seems to work fine now.

bkerler, 2024-12-09:
Seems the issue is the nvidia driver. Downgrading to 560.35.05-1 fixes the issues. According to other nvidia user forums, it seems that 565.57.01 has some bugs.

sl1pkn07, 2024-12-10:
tested with nvidia-565-77 and got the same

builded manually with https://github.com/prusa3d/PrusaSlicer/pull/13307 , https://github.com/prusa3d/PrusaSlicer/pull/13609 and https://github.com/prusa3d/PrusaSlicer/pull/13081 and some patches from gentoo (opencascade 7.8.0 patch and sed tweaks) deps with

```
  cmake -S PrusaSlicer/deps -B PrusaSlicer/deps/build \
    -G Ninja \
    -DDEP_WX_GTK3=ON \
    -DPrusaSlicer_deps_PACKAGE_EXCLUDES="JPEG;PNG;TIFF;MPFR;ZLIB;OpenSSL;CGAL;CURL;Cereal;EXPAT;OCCT" \
    -DPrusaSlicer_deps_PLATFORM_PACKAGES="JPEG;PNG;TIFF;MPFR;ZLIB;OpenSSL;CGAL;CURL;Cereal;EXPAT;OCCT" \
    -DPNG_LIBRARY_RELEASE=/usr/lib/libpng.so \
    -DZLIB_LIBRARY_RELEASE=/usr/lib/libz.so \
    -DPNG_PNG_INCLUDE_DIR=/usr/include \
    -DZLIB_INCLUDE_DIR=/usr/include
```

and prusaslicer iself with

```
  cmake -S PrusaSlicer -B build \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DSLIC3R_FHS=ON \
    -DSLIC3R_PCH=OFF \
    -DSLIC3R_STATIC=ON \
    -DSLIC3R_WX_STABLE=OFF \
    -DSLIC3R_GTK=3 \
    -DSLIC3R_BUILD_TESTS=OFF \
    -DCMAKE_PREFIX_PATH="${srcdir}/PrusaSlicer/deps/build/destdir/usr/local" \
    -DCMAKE_IGNORE_PATH=/usr/lib/cmake/wxWidgetsGTK/\;/lib/cmake/wxWidgetsGTK/\;/usr/lib/cmake/wxWidgets/\;/lib/cmake/wxWidgets/
```

works ok for me

[image]

greetings

SachCZ, 2025-03-04:
This should be solved in the latest 2.9.1-rc1. Please give it a try.

closed 2026-09-29 by prusaslicer-bot[bot]