prusa3d/PrusaSlicer #13619: Multiple beds with different print or filament settings

schl3nk3, 2024-11-21:
As starting to use the multiple bed feature in PrusaSlicer 2.9.0-alpha1 I was facing one missing feature. 

I would really like to give the single beds different settings. Especially when working on bigger projects with multi functional parts, some mixing of materials, print settings is typical.

**Describe the solution you'd like**
A .3mf project typically has one major set for print or filament settings. When working with multiple beds a change of one of these 2 settings, can be copied to all existing beds or will only affect the current selected bed.
In #10704 this feature was already mentioned by @strayr. 

**Describe how it would work**
Expecting working with multiple beds in the 3D space view. As soon, as the print or filament settings are change, a dialog could ask: "Should the change be used on all beds?". If not, a separate config should be saved for the current selected bed. All others remain. 

**Describe alternatives you've considered**
If the separation of settings to different beds isn't that easy. Maybe an enhanced modifier could help in many cases. So maybe it is possible to accept layer-height changes, filament-material selection for modifiers. I expect this to have a significant impact to slicing in total. But maybe when accepting the option sequential printing to be enabled it could get easier.

**Additional context**
If possible a listing of the separated bed configurations inside the objects (tree-view) on the right side would be great. So maybe it's possible to be changed there as well.
[image]

darraghbr, 2024-11-23:
#13617 Mentioned here as an extension of being able to 'name' beds.

BootsSiR, 2024-12-13:
Per bed settings would make this feature super useful.  I will often print various parts of a project with different settings (perimeters, supports, etc) and it would save a ton of time to be able to do it all in a single PrusaSlicer instance

5jvm0u4, 2025-03-30:
upvote

strayr, 2025-03-31:
Important point: why are we using (wanting?) multiple beds?

- project is to big to fit in one build plate

- some parts need to be a different colour, material or otherwise different setting

- job is being split amongst multiple printers. This might be a bit ambitious. 

It would be very easy to make something printfarm friendly that can split many parts up for a while heap of identical printers. But I think most domestic users do something very different, if they have multiple printers it's a collection with different capabilities, and wildly different sizes.

jhnphm, 2025-04-19:
This feature is needed to get around the 5-color limitation of the MMU/toolchanger. The workaround of separate 3mf files per material load/settings sort of defeats a lot of the utility of multiple build plates besides just larger print jobs.

cwagdev, 2025-05-24:
Just learned multi bed was added and this was the first thing I tried to do, would love to see it added.

msansjr, 2025-07-18:
As cwagdev said, it was the first thing I tried, since I am right now working on a project that uses different layer heights, speed and extrusion width for objects.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena