prusa3d/PrusaSlicer #13621: First use of 2.9.0 alpha 1 - slow panning in preview mode

KimmoHop, 2024-11-21:
### Description of the bug

Panning in preview with mouse (using right or middle button) is slower than in 3D Editor view, following mouse only about 50% of movement length. This is "slightly" annoying :)
2.8.1 works as usual, following mouse.

It seems this happened only on first use. Changing language "fixed" it and after closing and starting again it still works OK.
Maybe configuration from 2.8.x puts something off.
Or then it is Finnish translation ;)

(Rotation is something I'm never sure how it _should_ work, around the center of the screen? If an object is close (high zoom) it's easy to lose it from the view)

### Project file & How to reproduce

1. Install 2.9.0 alpha 1 **for the first time**
2. Start slicer
3. Use profiles from 2.8.x
4. Update configurations
5. Place object on bed. Pan to get the feeling.
6. Slice the object.
7. In preview pan moves **slower** and does not go full length.
8. It's slow also with MK4 input shaper machine and print profile
9. Change language or **restart** otherwise
10. Should work now :/

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Versio 2.9.0-alpha1

### Operating system

Windows 10

### Printer model

custom Voxelab Aquila (Ender 3 V2 clone)

Jan-Soustruznik, 2024-11-22:
Hello, @KimmoHop, thank you for your feedback. During internal testing, we noticed that it can sometimes be challenging to move the camera when all 9-beds are placed. I’m not sure if the issue you reported is related to this. Please let us know if the problem occurs again.

Thank you, and have a great day!

KimmoHop, 2024-11-22:
I was about to reply that I haven't seen it again. Tried even changing language back to Finnish ;)

Then, based on hunch or something similarly weird connection to your mention of plates, I added another plate, and _there we go again_... No need to clean install, just multiple plates. See how mouse and view are detached after second plate is added and sliced? Sorry for bad recording (amazed github will accept it) :)

[image]

KimmoHop, 2024-12-15:
Thanks, I can confirm panning multiple plates is fixed in **Version 2.9.0-beta1** 👍 

Will the **origin** be for each plate in the future, or will it always be located to plate 1? Not that it affects slicing etc but even _lightly OCD_ will notice it :D (Of course it should be separate issue, if it's not planned this way)
[image]

FidelCapo, 2024-12-16:
This issue behavior is hopefully improved in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo

Johnnl2004, 2025-12-04:
I have the same issue, just with the one plate. BambuLab does not show this phenomenon.
Running PrusaSlicer 2.9.4.
Is there any solution found?