prusa3d/PrusaSlicer #13622: 2.9.0 aplha 1 - Nozzle dips into previous walls with scarf seam

KimmoHop, 2024-11-21:
### Description of the bug

When **scarf seam** is set to _contours_ or _everywhere_, but **external perimeters first** is not set, nozzle will dip into previous perimeter when moving to and printing scarf section on external perimeter.
Nozzle tips are always wider than opening, more in V6 style, but also in narrow multi-color printer nozzles. (Image in [documentation](https://help.prusa3d.com/article/layers-and-perimeters_1748#recommended-thin-wall-thickness) shows quite unrealistic view of nozzle tip dimensions :) )

Should scarf seam set also external perimeters first, or be disabled when external perimeters first is unselected?
Should scarf seam be expert setting or external perimeter first normal setting? In normal mode user cannot see or set external perimeters first :/

In image, perimeters were printed at Z=26.05 and scarf section starts at Z=25.85. And no, I didn't try to print it: it might succeed, but it should leave a gap and ridge on middle perimeter. Or it might knock the print off.
[image]

### Project file & How to reproduce

1. Enable scarf seam
2. Don't enable external perimeters first
3. Slice object that shall have scarf seam (e.g. cylinder)
4. Nozzle path is _shown_ (one has to imagine realistic nozzle, path goes below top of previous wall while right next to it) to collide with previous wall
5. No warning is shown

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Versio 2.9.0-alpha1

### Operating system

Windows 10

### Printer model

custom Voxelab Aquila (Ender 3 V2 clone)

SachCZ, 2024-11-21:
Hello @KimmoHop, thank you for your feedback. I must admit I am not 100% clear on what you are describing as a bug. Can you please be more specific on the remedying action that we should take, or provide more detail on what is actually wrong? Please understand that **we cannot base our actions on a hunch that something might not work**.

Also, it would greatly help us if you provide an example of a failed print with a 3mf. But, if you do not feel like wasting filament, **do not feel pressured to do so.**

There is more to it than is apparent from your assessment. The move from the inner perimeter to the outer perimeter is indeed dipping into the internal perimeter. **But, it is the internal perimeter that has just been extruded**. In reality the extrusion does not stop perfectly at a spot and solidify instantly.

Then there is a gradual rise of the print head along an already printed and properly solidified perimeter, but one must not forget that the nozzle is quite hot and can deal with a slight overlap.

We did not see failed prints during our testing. **Do not get me wrong, our testing is of course limited and we might have missed cases in which the print might likely fail**. That is why an actual reproducible failing print would be very helpful.

KimmoHop, 2024-11-22:
This is the problem:
[image]

> action that we should take

Like I wrote:
Should scarf seam set also external perimeters first, or be disabled when external perimeters first is unselected?
Should scarf seam be expert setting or external perimeter first normal setting? In normal mode user cannot see or set external perimeters first :/

To do it right ;) I don't know how Orca deals with that, but it has e.g. inner-outer-inner sequence to keep **space clear** for exterior perimeter scarf.

> Also, it would greatly help us if you provide an example of a failed print with a 3mf. But, if you do not feel like wasting filament, **do not feel pressured to do so.**

I'll see if I can test this. 3 objects, 2 walls, no infill, no top layers - so one can see the result. You can test it too :) I didn't check if scarf is already in per-object setting.

1. scarf, inner walls first
2. scarf, outer wall first
3. normal seam

> **But, it is the internal perimeter that has just been extruded**. In reality the extrusion does not stop perfectly at a spot and solidify instantly.
> Then there is a gradual rise of the print head [...]

I'm a bit sceptic about these :/ Even if nozzle can deal with it, it pushes some of the other extrusion elsewhere and leaves a gap. Does the nozzle also collect filament (PETG?) to start a blob or casually drop it elsewhere?
If one were a certified welder, such gaps would be instant failure in x-ray control :D

SachCZ, 2024-11-22:
Hello @KimmoHop, I appreciate your enthusiasm. I have just printed what you suggested on my MK4. I do not see the problem you have theory crafted. Here is the 3mf and gcode: [scarf_test.zip](https://github.com/user-attachments/files/17871445/scarf_test.zip)

## Front view, from left to right:

1. scarf, inner walls first
2. scarf, outer wall first
3. normal seam
4. 
[image]

---

## Back view - meaning that **in the image on the left is number 3.**
[image]

There is quite a bit of missing material on the inner perimeter of number 2. "scarf, outer wall first". That is an **interesting** observation but it is not explained by your theory crafting, as **that is the case where the scarf is printed first.** The explanation most probably is the "unnecessary" move that happens between the end of the scarf seam and the start of the inner loop.

Please be so kind and add some evidence to your claims so we can take appropriate action. Currently I do not see a reason we should do any automatic setting management or display a warning.

Please do not get discouraged, as we are grateful for any feedback that can improve our product.

KimmoHop, 2024-11-22:
Final results are in, and you were mostly right, and my theory correct - _in theory_ :D

First, side view, seam or start of scarf seam at the bottom, left to right: normal seam outer first (a bit rough), normal seam inner first, scarf seam outer first (also a bit rough at start), scarf seam inner first. Scarf seam inner first **is the cleanest**, while scarf seam outer first looks really good on the inside. So very similar results 👍  My normal seams (or pressure advance) might need a bit tweaking ;)
[image]

Next, comparison between scarf seams from top to bottom: scarf seam outer first, scarf seam inner first.
[image]

Scarf seam inner first from slightly different angle:
[image]

The pit _is there_ (one point for me), though quite small (as would be expected from Bambu Lab A1 nozzle), but pretty much only where the nozzle landed. When the nozzle begins the movement, next wall sort of avoids it (not totally unlike Knight Bus in Harry Potter movies) or flows around it when speed rises. Against light, the beginning seems to have same thickness as normal wall, but could also be partly separated. I'm not sure if hobby knife is good enough to cut it for internal view.

_I_ would prefer _external perimeters first_ to have same visibility level as scarf seam settings, even though scarf seam itself would not be the reason. Nicer outer walls and more precise dimensions could be, for those of us not having Prusa (MK4s) printers :D 
_Obviously_ I have all settings visible ( I know enough not to change most of them ;) ) and can turn external perimeters on or off at will, so it's not important to _me_ in that sense.

Thanks for _digging in_ @SachCZ 👍 

This is how I printed the cylinders with my almost-least-favourite filament: 
[Scarf-cylinders.zip](https://github.com/user-attachments/files/17875297/Scarf-cylinders.zip)

SachCZ, 2024-11-22:
This was fun. I am glad we investigated. We will think about the settings levels and see what makes the most sense in the overall user experience so thanks for the feedback. I am closing this issue and I wish you happy printing :rocket: