prusa3d/PrusaSlicer #13623: 2.9.0-alpha1 incorrect calculation of volumetric flow rate

h4lf, 2024-11-21:
### Description of the bug

Incorrect calculation of volumetric flow rate for some combination of settings.

[image]
[image]
[image]

### Project file & How to reproduce

[bug36mm118.zip](https://github.com/user-attachments/files/17852691/bug36mm118.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-alpha1

### Operating system

Windows 10 Home 22H2 Build 19045.5131

### Printer model

Ender 3

Kaelum, 2024-11-22:
I was going to report under-extrusion, and other bugs, but they may all be related.  Infill doesn't contact the walls, and I could see that there is a significant reduction in extrusion as the infill gets closer to the wall.  This is all visible during printing, so it should be easy to diagnose.

jmd73, 2024-11-22:
@h4lf It seems to be related to the printer parameters "Use relative E distances" and "Retract on layer change".

If the checkbox "Retract on layer change" is checked, then you get a little bit more sane values, but they still seem to be off.
If "Use relative E distances" is checked the values are good.
If "Use relative E distances" is checked, the "Retract on layer change" parameter makes no difference to the result.

So the main culprit seems to be the checkbox for "Use relative E distances".

That is hopefully to some help for others to get them started on the investigation.

SachCZ, 2024-11-22:
Hello @h4lf, sorry for bothering you, but can you please clarify if it is just a visual bug or are there printing issues? What @Kaelum is suggesting is way more serious. @Kaelum, if you are seeing printing issues you did not see with the previous version (2.8.1) **please create a new issue**. For now we are treating this as a visual bug.

Also thank you @jmd73, your investigation is indeed of great use to us.

Jan-Soustruznik, 2024-11-22:
Hello, @h4lf , thank you for your report. We created internal issue, no. SPE-2577. We will try to fix in some next release. Have a nice day

h4lf, 2024-11-22:
> Hello @h4lf, sorry for bothering you, but can you please clarify if it is just a visual bug or are there printing issues? What @Kaelum is suggesting is way more serious. @Kaelum, if you are seeing [...]

I don't see any issues with printing.

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo