prusa3d/PrusaSlicer #13624: Crash detected

ttfranks952, 2024-11-21:
### Description of the bug

My Printer has detected a crash and keeps showing the screen. I have also lost the unload filament option. 

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Win 11

### Printer model

i3 MK3+

ttfranks952, 2024-11-21:
The screan keeps telling me to submit the error.

jmd73, 2024-11-22:
@ttfranks952 You should post this in the correct repo, this repo is for PrusaSlicer.
The correct place is [Prusa-Firmware](https://github.com/prusa3d/Prusa-Firmware)

Also post a image of the screen with the error present, also what firmware you are running on the printer.

closed 2024-11-22 by SachCZ