prusa3d/PrusaSlicer #13625: Ability to Cancel an Object prior to starting a print

BrienKing, 2024-11-21:
I currently have a object that I print 9 at a time, however sometimes I don't need that many and the number I need can vary from job to job.  

It would be nice to have a feature that allows me to cancel objects even before the print starts.

Scenarios might be 
I only need 3 of the 9 printed so I cancel 6 of them.
I only have X amount of filament left and I know I can get 4 done but not 9, so I could cancel 5 of them.
I only need 1 more (to make 10) so I cancel 8 of them.

This would keep me from having to slice 9 different versions and if any changes happen having to update 9 projects and re-slice them.

SachCZ, 2024-11-22:
Hello @BrienKing , is there a reason why starting the print a immediately canceling some of the objects does not work for you?

rcmaniac25, 2026-09-13:
Unless there's a spot for Connect (cloud and future local versions) requests, I would say this is still relevant on the PrusaSlicer side when doing send to printer.