prusa3d/PrusaSlicer #13626: M73 C commands aren't present

Woody-999, 2024-11-22:
### Description of the bug

When slicing multi color/material prints, the "M73 C" commands aren't generated. It's therefore still impossible to know when will the next tool change will appear. It seems to work when a pause is commanded in the slicer. 

### Project file & How to reproduce

[Multicolor print test.gcode.zip](https://github.com/user-attachments/files/17867522/Multicolor.print.test.gcode.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 Alpha 1

### Operating system

MAC / WINDOWS

### Printer model

MK4S

Jan-Soustruznik, 2024-11-22:
Hello,  @Woody-999 , could you share your 3MF file. thank you

Woody-999, 2024-11-22:
Of course, here it is

[M73_C_Test.3mf.zip](https://github.com/user-attachments/files/17868655/M73_C_Test.3mf.zip)

Woody-999, 2024-11-22:
Oh, ok. I didnt know that
Thanks for the repplies anyway

Jan-Soustruznik, 2024-11-22:
@Woody-999 , thank you for your feedback and the time you’ve taken to share it with us. If you notice anything amiss or have suggestions for improvement, please don’t hesitate to let us know. I am closing this ticket as resolved. Have a great day!

closed 2024-11-22 by Jan-Soustruznik