prusa3d/PrusaSlicer #13628: [docs] make it clear what the status & limitations of Crash Detection are, in docs/KB

m10d, 2024-11-22:
### Description of the bug

Crash detection was turned off in 5.0.0(?) for mk4, due to input shaping.  Then in 5.1.0(??) it was disabled by default for the XL, too, in favor of phase-stepping/VFA mitigation. 

This is obtusely hidden, visible only under `Calibrations` in the docs [ie this page](https://help.prusa3d.com/article/phase-stepping-xl_681760#phase-stepping-and-crash-detection)

**FIX:** the [official crash detection page](https://help.prusa3d.com/article/crash-detection_2100) doesn't say *anything* about this, and should.  Sure, the printer tags don't include the XL, but according to the phase-stepping calibrations page linked above, the XL _does_ have CD, just only if phase stepping is disabled?

I notice FOUR nearly identical questions about this in forums **today alone** :( it's clearly something the community needs a straightforward answer to.    There is abundant confusion between is this input-shapping related, VFA-related (or just stepper noise??), etc.  And many people just logically assume crash detection works.  Which it doesn't by default :( 

ETA: This issue is simply do **document** what the state is, so it doesn't require trying to search around totally irrelevant old info to guess the state; but I'll add that what I would really hope to see is some info doing a better job of summarizing the tradeoffs.  Surely Prusa engineers had very detailed testing showing "we can do CrashDetect, but only if we limit VFA mitigation to X", stuff along those lines.    What would I give up, besides noisier steppers, to re-enable crash detection?  Is re-enabling crash detection something Prusa is working on, `not-planned`, etc, etc?

### Project file & How to reproduce

read the forums and see how confusing this is to users

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

na

### Operating system

all

### Printer model

I own an mk4s, and and xl5t.

m10d, 2024-11-22:
just 30sec of searching:
- mk4s user thought there was crash detection https://forum.prusa3d.com/forum/postid/727068/
- xl user asks why crash detection doesn't work https://forum.prusa3d.com/forum/postid/727275/
- isn't collision detection supposed to catch this?? https://www.reddit.com/r/prusa3d/comments/1gx77a2/comment/lyflm8r/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button
- **top 1% commenter** in the sub sharing what is probably mis-information about Crash detection limitations https://www.reddit.com/r/prusa3d/comments/1gx77a2/comment/lyex8sq/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

hejllukas, 2024-11-22:
This is not related to PrusaSlicer but to the firmware. So I'm moving this issue to the Buddy Firmware repository.