prusa3d/PrusaSlicer #13629: Multiple Beds: Increase the limit

Souliboi, 2024-11-23:
**Is your feature request related to a problem? Please describe.**
Currently when trying to split up a project one appears to be limited to a total of 9 beds

**Describe the solution you'd like**
A setting in the preference pane to choose the limit as one pleases.

**Describe how it would work**
It feels relatively self explanatory, if a user desires and sees the need to split up a project in many smaller prints they could just increase the limit to a size they desire.

**Describe alternatives you've considered**
An alternative would be to make this a per project setting, which would potentially avoid the issue of some projects overwriting the preference pane of other users.

**Additional context**
Here's an example image of the limitation applying and not creating a 10th bed:
[image]

neophyl, 2024-11-23:
The release notes do mention the HARD limit of 9 beds though.  9 is a reasonable number given the layout choice and if more is needed then just split into different projects.

rcmaniac25, 2024-11-24:
I do agree it should be more then 9. Some of the creators I follow release pre-planned beds of parts, usually by color or by setting, and some going into 15 to 20 beds. Given the NxN value, I would say 16, 25, 36, etc. Better would be no limit (C++ does have types for dynamic resizing of arrays...). I have not looked at how it's handled in files, so it may be something there. But I feel like 9 is just a bit too limiting for some of the projects that exist at this point.

Souliboi, 2024-11-26:
> The release notes do mention the HARD limit of 9 beds though. 9 is a reasonable number given the layout choice and if more is needed then just split into different projects.

I did miss this portion of the release notes, my bad

closed 2024-11-26 by Souliboi

Tupson444, 2024-12-28:
Why close the issue? It is still a reasonable feature request (despite the "hard limit", which we don't even know what it exactly means, maybe it can be changed in future versions). I would like to have more than 9 plates (at least 25, if not unlimited).