prusa3d/PrusaSlicer #13631: Multi-Material Painter - Height Range Odd Orientation - 2.9.0-alpha1

BryanSmithDev, 2024-11-23:
### Description of the bug

Was going to test the new height range painter in the Multi-Material painter on the alpha and came across some strange orientation issues.

I expected the height range to go horizontally across the part but instead on this particular model it was going vertically and some other orientations. I have attached the project and a screenshot.

### Project file & How to reproduce

[image]

[PainterHeightRangeBug.zip](https://github.com/user-attachments/files/17882453/PainterHeightRangeBug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-alpha1

### Operating system

Windows 11 23H2 22631.4460

### Printer model

Prusa i3 MK4S

hejllukas, 2024-11-23:
Hi, thank you for your report. I was able to reproduce your issue. There is an issue with the Height range painting tool and rotated objects. I created an internal issue, no. SPE-2579.
We will fix it in the next release.

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo