prusa3d/PrusaSlicer #13632: Allow Color Changes and Pauses at Specific Tool X,Y Location

pausethelogic, 2024-11-23:
**Is your feature request related to a problem? Please describe.**
Currently, you can only add color changes (M600) and pauses (M601) on a specific layer in PrusaSlicer when looking at the sliced object. Since it's not possible to choose where a specific layer starts and stops, that also means that color changes and pauses can occur at any position on the object printing.

When doing things like pausing a print to insert a magnet, or doing single tool multi-material printing with manual color changes, this can cause unwanted artifacts on visible parts of the print during the resume purge, when instead we could force the printer to pause mid-layer at a specific X,Y location, such as inside infill or a specific part of the model.

Previously, I've seen the bottom slider was an indicator of the line number in the gcode file where that tool position happens, however as bgcode is now the default, making manual edits to the files isn't practical (and potentially not possible).

**Describe the solution you'd like**
A (seemingly) simple solution would be to add the ability to right click the tool path location slider (the bottom slider in the slicer preview window) and add a color change or pause, the same way you can with the layer height slider on the right side of the same preview.

This would allow us to control what X,Y position pauses and color changes occur, making manual MMU changes and intentional pauses much cleaner and more precise.

**Describe how it would work**
Without reviewing the code too closely, I imagine some code from [DoubleSliderForLayers.cpp](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/DoubleSliderForLayers.cpp) would need to be added to [DoubleSliderForGcode.cpp](https://github.com/prusa3d/PrusaSlicer/blob/ae97d00c34258f85b98fb000c131162863424a04/src/slic3r/GUI/DoubleSliderForGcode.cpp), as this seems to be the two GUI files for the vertical layer slider and horizontal tool path slider, respectively. 

Once the UI for the bottom slider is updated, a user can right click the bottom gcode slider and add a pause or color change, the same way you can with the vertical layer slider. PrusaSlicer would then insert the appropriate M600, M601, G0, G1, or any other commands to pause and resume the print at that X,Y position.

If we had the ability to force a wipe tower here too, that would be amazing too to help avoid color contamination. It could also allow users to make color changes pause and resume on a secondary object (similar to wipe towers being a unique model).

**Describe alternatives you've considered**
The only alternative I'm aware of would be to slice and export the project, convert the bgcode file to gcode, then find the correct line in the gcode file that correlates where you want the nozzle to pause, then inserting the gcode commands manually. Not ideal.

**Additional context**

Related issues this could fix:
https://github.com/prusa3d/PrusaSlicer/pull/13347

rtyr, 2024-11-23:
It looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5195.

pausethelogic, 2024-11-28:
> It looks like a duplicate of #5195.

It seems like a similar issue to #5195, but I can't find any acknowledgement from the Prusa team. Has anyone acknowledged this request? It seems to be wanted by the community

closed 2026-09-29 by prusaslicer-bot[bot]