prusa3d/PrusaSlicer #13635: 2.9.0-alpha1 - Toggle Printables Link

halofx, 2024-11-24:
PrusaSlicer has the option to disable downloads from Printables in the stable 2.8.1 release. The addition of a Printables link in the Main menu seems counter to a users decision to disable Printables downloads.

I would like to see either the Printables link automatically be removed if the Printables Download option is disabled, or the addition of a second preference to disable the Printables link independently of the Downloads preference. Much like the way the Log in for PrusaConnect can be disabled.

As someone that creates more of my own models, or uses models from private/client providers, this is just needless clutter. And annoying every time I click it by accident, which has been a lot in the last 2 days!

codefaux, 2024-11-29:
I second this. Emphatically.

I do not want an extra button to go to your partner's website. I don't use your partner's website, I use your slicer.

The Physical Printer tab is invasive enough but at least it's justified. It uses enough system resources on a slim system to be problematic when accidentally clicked, and even on a quick system it's noticably slow to come up because it's an _external website._ That's what web browsers are for, and your WebView doesn't respect adblock, which is probably a big part of why it's there, nor do I know which tookit you've added to my system allowing for potential security issues.

The app is turning into a mine field. It's the super-early days, but if we don't raise our voices now when "it's just one/two buttons" it'll definitely get worse. I don't want a slicer with eight buttons to load the four things I want and four things _you_ want.

I believe my stance is likely to be widely supported in your userbase.

Please include an option to toggle ALL WebView-associated features OFF and remove their buttons.

halofx, 2024-11-29:
> nor do I know which tookit you've added to my system allowing for potential security issues.

On Windows it is using Microsoft Edge Webview, you can see this in the Task Manager, the views are clustered with PrusaSlicer. I believe I remember hearing this is a mandated requirement of the OS, and would expect it to be Safari on a Mac.

sithlord48, 2024-12-21:
> I second this. Emphatically.
>
> I do not want an extra button to go to your partner's website. I don't use your partner's website, I use your slicer.
>
> The Physical Printer tab is invasive enough but [...]

I could not agree more. This is not the way to compete with them.

Tupson444, 2024-12-21:
I too agree with this. I only rarely use Printables, and when I do I prefer using a regular browser. I don't like it being the same style and right next to the setting tabs (Print, Filament and Printer). To someone it might be the perfect position (e.g. if primarily downloading files from Printables, then slicing), but not everyone. I would like it to be at least separated from the other buttons, as it is a completely different thing from the setting tabs.

Of course, there is no need to make one configuration fit all, simply add the option to toggle it off. 

(It would be good to always add a toggle option for all new things that are not a necessary part of the software, no need to wait for user feedback to realize that, unsurprisingly, many don't like it.)

CRCinAU, 2024-12-27:
Its this same stuff that caused them issues with the AppImage - which has caused them now to move to flatpak and install about *an entire gigabyte* of dependencies.

Same with the fedora packages:
```
Transaction Summary:
 Installing:       105 packages

Total size of inbound packages is 867 MiB. Need to download 867 MiB.
After this operation, 1 GiB extra will be used (install 1 GiB, remove 0 B).
Is this ok [y/N]:
```

There really should be a way to compile this out at package time for those who don't want to install a gigabyte of bloat so that Prusa can push their own web site.

codefaux, 2024-12-27:
This seems worth forking. I absolutely don't have unpaid time to dedicate to this, but if there's a bounty I'll contribute.

SparkyBluefang, 2025-01-01:
Relevant #12968

jltx1, 2025-01-02:
I agree with the others that the placement of the "Printables" button is just bad.  It is easy to hit accidentally (has "print" in the name) and doesn't make sense in the workflow.  The "Physical Printer" makes sense so not sure why that is getting any grief. 

As a counterpoint, however, I DO LIKE the access to Printables from within the slicer where I can click a button and the object is on the platter ready to slice.  I will use it rarely but I prefer the interaction over using a web browser and downloading / managing files, though having both options is great.   

**Proposal:**
Orca slicer has a Home menu that shows recent projects and such which is really smart.  This is the perfect place to add a portal to Printables.   You can move the app settings and account under here too.  In the spirit of open source you should consider adopting that idea because it's just better.   They have other good interface ideas that you should adopt as well.

codefaux, 2025-01-02:
I don't agree. I don't think the solution is to fundamentally change the workflow of the application. That also doesn't address any of the actual concerns raised aside from button placement.

You can have a button to click to put a Printables object sliced on the plate from a normal web browser on your system, the slicer doesn't need to provide that by also hiding a web browser under its cloak.

The physical printer button is getting grief for the reasons stated in the messages, if you read them you'll be sure.

It sounds like you enjoy the features provided by Orca. Most of the new features here have come from other slicers who are focused on community ideas and printing features instead of adding an ice maker and a moving walkway.

synthead, 2025-01-03:
I am interested in this feature.  Moreover, I would prefer Prusa to not make any external web requests at all.

jltx1, 2025-01-03:
To be clear, I support your desire to have a browser-free slicer.  I'm just pointing out that while I agree the current implementation is bad, I hope they come up with a good compromise.  If someone had asked me if I wanted that feature I would have said no (maybe hell no) but after trying it I like it ok, even though 99.9% of what I print is my own designs.   But it is a slippery slope.  Next the Jo Prusa paperclip will become a chatbot.

codefaux, 2025-01-03:
@jltx1 -- I understand your stance, and I'm not trying to engage in an internet argument, but I'd like to clarify that external browser integration provides for externally sourced objects from partnered sites, and anything else is explicitly what we're objecting to.

Any internal integration with a website requires a browser in the app.

I am (we are) against having a browser in the app, full stop, no exception.

You cannot say you support having a browser-free slicer while suggesting they compromise on not including a browser so you can have your external integration inside the slicer.

If you like the models available to download and print, go to their webpage with your browser of choice and slice them with your slicer of choice.

If you enjoy being limited which options are the absolute most convenient, that's your business, but if you remove the in-slicer integration they're all equally convenient and I guarantee different sites have different content, so you'll actually be better suited by your lack of laziness by firing up that browser you already have, instead of duct taping one -- which only goes to one place -- to the side of a slicer.

The compromise beteween the two is the explicit part we're objecting to. There is no middle grounds. I appreciate your attempt at support, but you are not supporting our intent. Our intent is counter to your stated interests.

kubispe1, 2025-01-06:
Hi, thank you for your discussion and feedback. We appreciate the time and effort the community puts into sharing their perspectives, and I would like to take this opportunity to clarify our goals and focus regarding the "Printables" tab introduced in version 2.9.0.

We understand that some users are not entirely satisfied with this new feature. While we have received significant positive feedback highlighting the convenience and workflow enhancements it offers, we also recognize that not all changes will resonate with every user. The Printables community is extensive, and we are confident that many will find value in this integration over time.

It’s important to note that this is the first iteration of the "Printables" tab. Our initial priority has been delivering core functionality, and as we move forward, we are committed to refining the feature. This includes addressing user concerns, enhancing usability, and ensuring a smoother, bug-free experience. However, our focus is on overall improvement rather than making every individual button, icon, or tab fully customizable.

Thank you for your understanding.

closed 2025-01-06 by kubispe1

codefaux, 2025-01-06:
We are not asking for every button to be customizable. Being intentionally reductive is inflammatory and naive.

I doubt this was discussed with more than one or two developers, and I'm betting it was a corporate decision and not an open source one.

This is not completed. Tag it as won't fix and be honest with your consumers.

It's official; I'm not using PrusaSlicer anymore.

sithlord48, 2025-01-06:
> Thank you for your understanding.

Understanding... I think we are starting to understand its not about open source and what the community wants.. 

Please remark this won't fix because this issue has not be resolved and you literally just told it its wont be fixed.

Tupson444, 2025-01-06:
>... rather than making every individual button, icon, or tab fully customizable

This wouldn't be asked if the button placement and appearance wasn't badly chosen in the first place. We are all giving valuable feedback how to improve it. The fact that other users didn't complain doesn't mean they too wouldn't benefit from better button placement.

halofx, 2025-01-07:
> Thank you for your understanding.
I am not really feeling any understanding in this reply, it rings very tone deaf to me. Prusa saw the need to make the login optional, and the need to make the Open in Slicer option on Printables optional. How does this not carry as much weight? The location is horrible. 

A few days ago, I accidently clicked the link yet again, and in the middle of cursing, I realized I actually did need a file from a collection I had. But then realized that I was automatically signed into Printables, but with the wrong account, and no way to change it. My Prusa3D account used with Connect is my business email, created years before Printables. My primary Printables account is a community/media email. So I can't even use the Printables integration the one time I actually wanted to try it.

I acknowledge that many printer owners are routinely looking for things to print, in an effort to justify having a printer. But I haven't found myself in that boat in many years. 

To be clear, as a business user, and owner of many Prusa machines, my files are either produced by me, or customer provided. I don't see anytime when this feature will ever justify its existence for business use. So the only singular button I want to fully customize is the singular binary setting of OFF, tied to the already existing preference Allow downloads from Printables.com. That is it. No extra UI needed.

Tupson444, 2025-01-07:
I understand the aim to avoid too must customizability, but I think this is still a valid request and shouldn't be rejected, because it's just a simple toggle in the settings.

As an alternative, I've opened a separate issue asking to relocate or change style of `Printables` tab (which should be done in any case, even if adding a toggle).

codefaux, 2025-01-10:
> it's just a simple toggle in the settings

Every new setting is a single new setting. I'm on your side, but that's a really weak argument.

Side note: The Flatpak freezes and crashes on Linux any time I accidentally click either the Printables or Physical Printer tab, so that's super awesome. If only my slicer didn't also have a blender and a can opener tied to it.

30350n, 2025-01-17:
> However, our focus is on overall improvement rather than making every individual button, icon, or tab fully customizable.

When the "Log In" button in the top-right was first introduced, I could immediately disable it in the settings. There is **no** reason why this wouldn't be possible for the new "Printables" tab.

The way it is placed right now, it actively disturbs my (and probably many other users) workflow. I almost exclusively print my own models or modified versions of models found online and constantly use all other tabs from the top bar to adjust settings, etc. 
I don't use (nor want to use) the Printables Tab.

Please reopen this.

Gallion, 2025-01-29:
I want to voice my displeasure about the existence of that tab too.

TomaszSzt, 2025-04-10:
Please add option to hide it in GUI configuration. It shouldn't be that hard. I am using this software at work for processing internal projects so it is at least counter-productive. It may also introduce security issues as a part of what looks like an application is in fact a web content. Users are usually wary when accessing / downloading things through the web-browser, but when it is provided as a part of a program this awareness drops. And the content of "printables" is user generated and may contain malicious data.

codefaux, 2025-04-10:
> introduce security issues

Even a confidently security-aware user faces security issues with this empty trojan horse kit. The user has ZERO exposure to security certificates or confirmation, they have no manner of assurance that their web-browswer-slicer-salad-shooter-lint-comb has not been compromised. Zero assurance that the target site hasn't been hijacked and/or poisoned. Zero assurance that the login they're about to enter isn't fishing.

"Hey you guys remember how web browswers made this HUGE DEAL of how the security icon on the toolbar was like, really important to understand and see? Let's just remove that."

Furthermore, it CONTINUES to crash every time I accidentally click it, even if I click away before the page finishes loading. Because it's a buggy nightmare, despite using Flatpak to distribute it? I thought the point of Flatpak was that it would be the same software on everyone's system since it distributes a copy of literally every dependancy?

What was wrong with having a slicer that was just a slicer? It didn't provide an auxiliary revenue stream. Do the team of coders love chasing bugs in a WEB BROWSER instead of writing a slicer? Unlikely. F the intelligent/forward-thinking users, grandma clicks ads and the C suite loves it.  Carry on.

GWdd, 2025-08-26:
Sooo ... I agree with just about every negative listed about a rogue web browser on my PC.  

How do I remove all of the Prusa/Printables credentials I muddlebrainly entered into PrusaSlicer so it doesn't accidentally expose private information?

codefaux, 2025-09-01:
What's worse is that it's not even stable. It crashes every time I accidentally open it, and I'm using the fkn Flatpak because they decided normal package managers are too hard. So I'm forced to use a Flatpak (which restricts where I can put my files if I want to print them!) BECAUSE Prusa feels the need to force a rogue browser on its users.

I also would be interested in purging the included malware's datafiles. Where can they be found?

codefaux, 2025-09-01:
See also https://github.com/prusa3d/PrusaSlicer/issues/12968#issuecomment-3014225249 for an Arch pkgfile (and earlier for other options) of a version with the in-app malware browser stripped out.