prusa3d/PrusaSlicer #13637: 2.9.0-alpha1 - part cutting improper behavious

EricPobot, 2024-11-24:
### Description of the bug

When cutting a part on an additional bed, the new parts are created on the default one and thus must be moved afterwards.

The logical behavior would have been to create the parts on the same bed the original one was located.

### Project file & How to reproduce

1. Put a part on an additional bed
2. Apply a cut to it

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-alpha1

### Operating system

Mint 21.3

### Printer model

i3 MK3S

Jan-Soustruznik, 2024-11-25:
Hello, @EricPobot , Thank you for your featback. We created insternal issue no. SPE-2583. We will try to fix in some next release. Thank you

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo

EricPobot, 2024-12-16:
Works fine in latest beta. Thanks a lot for fixing this.