prusa3d/PrusaSlicer #13640: 2.9.0-alpha1 Fuzzy Skin and random seam do not combind

CableCatDK, 2024-11-24:
### Description of the bug

When using the new paint-on fuzzy skin feature and random seam together. The part which have paint-on fuzzy skin, do not have random seams. 

The root cause is probaly the same as in SPE-2576.

[image]

### Project file & How to reproduce

[fuzzyskin-bugreport.zip](https://github.com/user-attachments/files/17892938/fuzzyskin-bugreport.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0-alpha1

### Operating system

Windows 11 - 10.0.22631.4460

### Printer model

MK4 + MMU3

Zohan-4775, 2024-11-29:
[image]
Same Problem for me, only with painted Fuzzy Skin

tjstansell, 2024-12-08:
I tried adjusting the seam settings on my object with painted fuzzy skin and none of the settings changed the outcome, so it seems to ignore all seam selection when you paint the fuzzy skin.

CableCatDK, 2024-12-19:
The issue is fixed in RC1

lukasmatena, 2024-12-19:
Correct, this is a duplicate of #13640. Closing.