prusa3d/PrusaSlicer #13643: Prusa Slicer 2.9.0 A filling in hollow models

csweeney0926, 2024-11-24:
### Description of the bug

I was reprinting/reslicing a MK4 fan duct (my original one got a little melty after long use) and 2.9.0 Alpha seems to be trying to make the largely hollow print solid when slicing. Not sure why this is happening the model is manifold and using "the fix by windows repair algorithm" actually makes it worse. I was able to slice it just fine in older versions of Prusa Slicer but it looks like there are some kinks to work out in 2.9. I uploaded the file in question for reference. Interestingly the problem seems to be intermittent. I reopened it and resliced it and it worked this time curious...
[Mk4 Improved Fan Duct V3.1.zip](https://github.com/user-attachments/files/17893891/Mk4.Improved.Fan.Duct.V3.1.zip)

### Project file & How to reproduce

[Printable parts mk4 github.zip](https://github.com/user-attachments/files/17893915/Printable.parts.mk4.github.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 Alpha

### Operating system

Windows 11 64-bit operating system

### Printer model

MK4

csweeney0926, 2024-11-25:
Unfortunately, I didn't save the g-code with the issue. I figured it was something wrong with the slicer so I closed it and resliced it with 2.8 (which worked fine). Then I re-opened 2.9.0 A to try slicing it again (I wanted the scarf seams) and it worked the 2nd time. So I guess the problem is intermittent or potentially a one-off.

Jan-Soustruznik, 2024-11-28:
Hi @csweeney0926, Thank you for your response. We were unable to replicate the issue you mentioned. If you encounter the same problem again, please save and send us the 3MF and G-code files (or any other relevant output), especially if you're using an Alpha or Beta version. These versions are specifically release to public to help us debug and resolve reported issues before releasing a stable version.

For now, I will close this issue. If you experience the problem again or are able to provide the 3MF and G-code files, please feel free to open a new issue.Thank you for your help!

closed 2024-11-28 by Jan-Soustruznik