prusa3d/PrusaSlicer #13646: Add Capability to open Bambu Studio 3MFs

csweeney0926, 2024-11-24:
I occasionally get requests to print items that are posted on Maker World (Bambu's version of Pintables).  For some files a stl is not available and only a 3mf or Bambu studio file can be downloaded. Prusa slicer is not able to open 3MF files from Bambu Studio (they added some feature that isn't compatible with the current Prusa Slicer Versions. For these files I have to open them in Bambu Studio, export as an STL and then import the STL into Prusa Slicer which is a lot of extra steps. 

It would be nice if Prusa Slicer could open or convert Bambu Studio 3MFs into STLs directly. As Maker World gets a larger library and more people use it I expect this problem will pop up for more people more often.

raistlinJ, 2025-08-09:
I created a converter here: https://github.com/raistlinJ/3mf_bambu2prusa
Tested with several models and it seems to work; even transfers painted objects.
Hope it helps.

thomergil, 2026-01-17:
@raistlinJ , the tool is great but requires Python-Tk. Why not make this a command line tool to make this easier?

timur-tabi, 2026-09-19:
I would still like to see this feature in PrusaSlicer 3.0.