prusa3d/PrusaSlicer #13648: Problem with the amount of filament used.

bujias02, 2024-11-24:
### Description of the bug

In the latest stable version of "PrusaSlicer," for a specific part, the filament usage calculation is around 60g. However, in this new version 2.9.0 alpha1, the filament usage calculation exceeds 100g, while the actual weight of the part is 60g.
[image]

### Project file & How to reproduce

Slice any part.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.9.0-alpha1-win64-202411202012

### Operating system

windows 10

### Printer model

Anet A8

Jan-Soustruznik, 2024-11-25:
Hello, @bujias02 , Could you share your project file(3MF)? Thank you

Jan-Soustruznik, 2024-11-28:
Hello @bujias02,Thank you for your feedback. We have identified that the issue is caused by disabling the "Use Relative E Distance" setting. We will try to resolving this in one of the upcoming releases. Internal issue has been created: SPE-2593.

Thank you for your support!

bujias02, 2024-11-28:
It's great to have a team that works so well.

Thank you so much.

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo