prusa3d/PrusaSlicer #13649: 2.9.0-alpha1 Wipe Tower does not auto disable on a single filament print

seifer69er, 2024-11-24:
### Description of the bug

With a MK4s MMU3 setup and using a single filament print,
on previous releases slicing the print would auto disable wipe tower on the platter.
in the current 2.0.0-alpha, the doing the same steps, the wipe tower does not auto disable and has to be deselected via the profiles
This is a very annoying bug for single filament prints

### Project file & How to reproduce

MK4s MMU3 default profiles,
slice object and the wipe tower will auto align in the platter

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-alpha

### Operating system

Windows 11

### Printer model

MK4s MMU3

seifer69er, 2024-11-25:
Recommend clicking on the wipe tower and disabling it when pressing delete on the keyboard

FidelCapo, 2024-11-25:
Thank you for the report. I can confirm the reported issue, and it is likely the same as the one described in issue https://github.com/prusa3d/PrusaSlicer/issues/13650. We will work on fixing it.
SPE-2580 created

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo

seifer69er, 2024-12-16:
Can we please have the feature added where we click on the wipe tower and can delete it by pressing the delete key on a keyboard, this would save a lot of hassle