prusa3d/PrusaSlicer #13650: 2.9.0-alpha1 arrange on xl single material doesn't arrange to front

pipcorona, 2024-11-25:
### Description of the bug

When arranging on a Prusa XL and using only a single extruder, the models no longer arrange at the front corner of the build plate to limit the amount of heated section that need to be used and arrange at the back of the build plate as if there was a purge tower still. I tried with the Arrange button and the Arrange Current Bed button. 

### Project file & How to reproduce

Put any model on the bed, print with a single extruder only, arrange

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-alpha1

### Operating system

MacOS 15.2

### Printer model

Prusa XL 5 Head

FidelCapo, 2024-11-25:
Thank you for the report. I can confirm the reported issue, and it is likely the same as the one described in issue https://github.com/prusa3d/PrusaSlicer/issues/13649. We will work on fixing it.
SPE-2580 created

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo