prusa3d/PrusaSlicer #13651: Model does not stay in the same place of the bed when using multiple beds.

DavidBjerreBjoerklund, 2024-11-25:
### Description of the bug

When using the new multiple beds feature, I center the model on the bed and a similar model on an extra bed and then saves the project. When I reopen the model on the extra bed is no longer centered but outside printable area.
[image]

### Project file & How to reproduce

[Komprimeret arkiv.zip](https://github.com/user-attachments/files/17897994/Komprimeret.arkiv.zip)
Open the file. Replace model on the bed to the left and press save. Then when reopened it will decenter the model again, at least with my machine/slicer combo.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.9.0 alpha1

### Operating system

macOs 15.1.1 (24B91)

### Printer model

mk4s

FidelCapo, 2024-11-25:
Thank you for the report. I can confirm the reported issue. We will work on fixing it.
SPE-2581 created

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo