prusa3d/PrusaSlicer #13653: 2.9.0-alpha1 - AppImage

probonopd, 2024-11-25:
The [release notes](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1) for PrusaSlicer 2.8.1 state

> PrusaSlicer now depends on WebKit library, which greatly complicates its distribution. (...) The AppImage made sense when it could be used in the "bundle what you need, distribute a single file" way, but having to distribute several different AppImages and maintaining the required build infrastructure (and still worrying about what needs to be updated when some Linux distribution update is released) means burning time that we would much rather invest into actual work on PrusaSlicer.

This can be achieved by bundling WebKit inside the AppImage (as I have done, see below).

According to the [release notes](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) for PrusaSlicer 2.9.0-alpha1, Prusa is considering to move away from the AppImage format for Linux, and is anticipating that this change might not be welcomed by all users. According to the release notes, Prusa would want a PrusaSlicer AppImage to:

* Bundle all necessary dependencies, including libraries such as glibc or webkit. - ✅ **DONE** by bundling _all_ dependencies of the application rather than _just the depencencies that cannot be reasonably assumed to be part of each target system_
* Not require specific versions of system libraries, such as libfuse, to be present on the target system. - ✅ **DONE** by using a static AppImage runtime that does not use libfuse from the system, and bundling _all_ dependencies of the application
* Be able to run on multiple Linux distributions without requiring extensive maintenance. - ✅ **DONE** by the above
* Allow for easy building and distribution of multiple architectures, such as x86_64 and aarch64. - ✅ **DONE** by using tooling that supports these architectures (e.g., Docker)
* Not require building on older distributions for compatibility reasons. - ✅ **DONE** by bundling _all_ dependencies of the application
* Provide a seamless user experience, including desktop integration and application updates. - (☑️) **OPTIONAL** using an optional [desktop integration](https://github.com/AppImageCommunity/awesome-appimage?tab=readme-ov-file#desktop-integration) tool (but not desirable in all cases, see below)
* Offer a sandboxing mechanism to control application permissions. - (☑️) **OPTIONAL** using a third party sandbox (but not desirable in all cases, see below)

__tl;dr__: By using a static runtime and by bundling _all_ dependencies inside the AppImage, the stated goals can be achieved at the cost of a larger download size (although not as extreme as Flatpak where all of GNOME gets downloaded even on non-GNOME systems).

### Version for testing

https://github.com/probonopd/PrusaSlicer/releases

So far I have tested the AppImage on
* Windows 11 with WSL2 in a Debian environment ✅
* Ubuntu Studio 20.04.3 LTS (Focal Fossa) on hardware with Intel graphics ✅
* Fedora Linux 38 (KDE Plasma) on hardware with Intel graphics ✅
* Fedora Workstation 40 (GNOME) on VirtualBox ✅
* Manjaro Linux XFCE 24.1.2-241104 (like Arch Linux) on VirtualBox ✅
* Chimera Linux 20241027 (which uses musl libc instead of glibc and doesn't use many parts of GNU) on VirtualBox ✅
* NixOS 24.05.6668 (KDE Plasma) (which doesn't have `/usr` populated) on VirtualBox ✅
* Ubuntu 16.04.7 LTS (Xenial Xerus) (8 years old) on VirtualBox (☑️) with minor graphical glitches

Please let me know your experience with using this AppImage, mentioning your Linux distribution and version.

probonopd, 2024-11-25:
### Details

Prusa Research has been offering PrusaSlicer for Windows, Mac, and Linux from their own download page and from GiHub Actions for quite some while. For Linux, Prusa has been using the AppImage format for years, as it provides a way to have one convenient download for various Linux distributions. This has has been working well for many happy users over the years. 

However, the [release notes](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) for PrusaSlicer 2.9.0-alpha1 state that Flatpak will be used as an official way to distribute PrusaSlicer for Linux in the future.
While I fully respect Prusa Research's freedom to distribute their software in every way they wish (after all, this is the main point of the AppImage project), the release notes make some claims about the AppImage format that require a closer look because they are not (or no longer) factually accurate:

* **It is claimed that AppImage is not designed to bundle "everything" and requires assumptions about the target system.** AppImage, first and foremost, is a self-mounting disk image that merely mounts itself and runs whatever the aplication author has put inside. In that regard, it is not unlike a `.zip` file. In deciding how much to bundle privately vs. how much to use from the target system, an application author is at liberty to make a tradeoff decision between size and robustness. Traditionally, most application developers wanted to bundle only those dependencies of an application _that could not safely be assumed to come with all supported mainstream desktop Linux distributions_. Since Linux distributions apparently are [not reaching universal consensus](https://gitlab.com/probono/platformissues) on how to do even basic things (e.g., in which location to put certain files), it is not easy to find a "common ground" that can be assumed to be part of every supported target system. A common-sense approach that has worked reasonably well for many is to target the oldest still-supported Ubuntu LTS release. **However**, this is not the only one to do things, and the application developer might also decide to bundle _all dependencies of an application, full stop_.
* **It is claimed that AppImage requires manual decision-making and testing on all targeted Linux distributions.** While true, that decision can very well be to bundle _everything_ the application needs to run. This will result in a larger download size (although not as extreme as with Flatpak, as explained below).
* **It is claimed that that some libraries may be "almost or completely" impossible to bundle (e.g., glibc or webkit).** Tools like go-appimage or sharun exist that entirely automate the handling of glibc. As for webkit2gtk, some manual steps are currently needed because that software is not built with relocatability in the filesystem in mind (in other words, it uses compiled-in hardcoded paths). However, this can be solved by some trivial patching (or submitting an upstream fix).
* **It is claimed that AppImage requires a specific version of libfuse to be present on the target system.** While this used to be the case with older versions of the AppImage runtime (a tiny piece of software that is part of evey AppImage - not to be confused with the large Flatpak runtimes), it is no longer the case when you use a [static runtime](https://github.com/AppImage/type2-runtime) that does not rely on any libraries from the target system, including libfuse. In fact, today there are several static runtiems from different authors, one even written in Rust. For systems with no FUSE support at all, an AppImage can be extracted (similar to a `.zip` file) using the `--appimage-extract` command line option. For this, no FUSE is needed at all.
* **It is claimed that Using old glibc for compatibility reasons forces building on older distros.** This is not an inherent limitation of AppImage. It is merely a consequence of application developers wanting to privately bundle only the dependencies that _cannot reasonably assumed to be part of every mainstream Linux distribution_. When you develop against an OS, then usually your users will need that OS version or a newer version in order to run your application. The only way around this is to _bundle everything_ - which is what Flatpak does, and is what an application author can choose to do with AppImage as well.
* **It is claimed that Flatpak solves the dependency hell problem by building against a defined runtime.** While true, this means that on top of the user's Linux distribution, essentially the whole stack is installed a second time (e.g., in the case of PrusaSlicer, all of GNOME even if you are not using GNOME at all). And the runtime is definded by _someone_, not by the application author. There are some applications that require certain versions of (patched) dependencies. This is easy with AppImage but not with Flatpak.
* **It is claimed that Flatpak provides better desktop integration and application updates.** Depending on the use case, this may be an advantage or a disadvantage. For example, users wanting to try out the latest alpha version don't neessarily want that version to be integrated into the desktop, and certainly don't want the previous alpha to go away when a newer one is downloaded. With AppImage, users can have as many versions around as they like to. For example, having an AppImage would be especially be beneficial for each of the development branches in the PrusaSlicer project [... cut ...]
Luckily, it's not either-or: AppImage and Flatpak serve different purposes for different target groups (portable single-file applications vs. installed applications). So why not have both?

References

* https://github.com/prusa3d/PrusaSlicer/issues/13376 
* https://github.com/prusa3d/PrusaSlicer/issues/13361 (no more need for two separate AppImages)
* https://github.com/prusa3d/PrusaSlicer/issues/12984 (libwebkit2gtk-4.0 is now bundled)
* https://github.com/prusa3d/PrusaSlicer/issues/12835 (libwebkit2gtk-4.0 is now bundled)
* https://github.com/prusa3d/PrusaSlicer/issues/12922 (_all_ libraries are now bundled)

Samueru-sama, 2024-11-26:
I want to add this to show how huge flatpak download sizes are: 

[image]

Flatpak with deduplication and all of that ended up using **9.85 GiB**, while the same apps as AppImage used **2.0 GiB**. And I was not able to find all the apps that I have appimages of as flatpak, like lite-xl or deadbeef.

buhman, 2024-11-26:
https://github.com/probonopd/PrusaSlicer/releases/tag/mk1 works for me on Gentoo (`default/linux/amd64/23.0/desktop/gnome` profile).

Souliboi, 2024-11-26:
Honestly, the replacement of AppImage to Flatpak has been a long time coming,
imagine you are some individual, just wanting to download a slicer and being confronted with a table on what is, and what isn't capable of running the release binaries, and then hoping nothing goes wrong in the process.
Flatpak takes away any and all of the worries and makes it as simple as just clicking install, removing any of the prexisting efforts to explain the individual user what they should, and shouldn't download.

> Choosing a Flatpak bloats up the size too much

First, it's not the early 2000s anymore, we can spare a couple gigs of space
Second, this assumes that modern day users don't already use some sort of Flatpak, which they are **very** likely to do so, which in turn means this "inflation" by downloading the GNOME Runtime doesn't really mean anything here, it would be present anyway.

> Just include all dependencies

Then we would just end up inflating the package now would we, otherwise we would just end up requiring to test for any individual distro out there and pray they all play nice.

> Flatpak forces sandboxing, AppImages don't

Security by default is a good option, yes, that includes sandboxing a slicer

And to finish this comment of mine off, anytime I had involvement with AppImages, it has always been dreadful to work with them if they didn't feel like coorperating with me and didn't "just work", this was even the case with PrusaSlicer, where thankfully, most of this for me could be avoided by 
A: Installing the Flatpak
B: Installing the Distro's package, which also depends on the distro keeping it up to date.

It has always been a headache universally, pretty much regardless of what AppImage I came across, you could blame it on the packager or the user, but at this point I couldn't be bothered.
Signed, a Linux user frustrated with AppImages

Samueru-sama, 2024-11-26:
>Then we would just end up inflating the package now would we, otherwise we would just end up requiring to test for any individual distro out there and pray they all play nice.

Nope, you can see it in my comparison, some of the AppImages I have there (like the Steam appimage) already bundle everything and work on any distro.

I recently made this appimage of [cromite](https://github.com/pkgforge-dev/Cromite-AppImage) that bundles everything and works on any distro, and **the appimage is smaller than the tar.gz that cromite offers for linux** which isn't really portable as it depends on glibc.

The difference here being that we bundle everything we need and no more. The flatpak runtimes are more like taking a AppImage and using distrobox to spin a container to run them that's compatible with it.

Anyways, none of this matters like you said, however **if one option will use many times less storage than the other and gives me the same thing**, I will pick the one that uses less storage.

>Security by default is a good option, yes, that includes sandboxing a slicer

I do have a problem with not having a way to disable it, even snap provides a method to disable the sandbox, because not all apps play nicely with it. Also in the case of firefox browsers, [the sandbox is bad for its security](https://librewolf.net/installation/linux/#security). The cromite the dev doesn't want to provide a flatpak of it because of these issues. 

With that said I'm not sure what the implications could with in the case of PrusaSlicer, no idea if webkitgtk even has the ability to use namespaces to create its own sandbox.

>anytime I had involvement with AppImages, it has always been dreadful to work with them

Sad to hear, and I get it I also know some appimages that broken.

My story flatpak isn't great though, mine was short and a nightmare with yuzu being broken as flatpak because [of the outdated mesa they shipped](https://www.reddit.com/r/yuzu/comments/11307f0/comment/j8o6gsa/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) which was out of their control, then I found out how much storage I was wasting with all the flatpak runtimes I had and just stopped bothering with it.

>It has always been a headache universally, pretty much regardless of what AppImage I came across, you could blame it on the packager or the user, but at this point I couldn't be bothered.
Signed, a Linux user frustrated with AppImages

In any case, please note that the AppImage that probono linked fixes these common issues for good.

probonopd, 2024-12-02:
@madpipeline I am trying to address the issue you mentioned [there](https://github.com/prusa3d/PrusaSlicer/issues/13675#issuecomment-2509980483).
Which language is your system set to?

probonopd, 2024-12-02:
Thanks @Samueru-sama for bringing my attention to this.

Currently PrusaSlicer installs as part of `make install`

```
 /usr/resources/localization/ko
 /usr/resources/localization/ko/PrusaSlicer_ko_KR.po
 /usr/resources/localization/ko/PrusaSlicer.mo
 /usr/resources/localization/ja/PrusaSlicer_ja.po
 /usr/resources/localization/ja/PrusaSlicer.mo
 /usr/resources/localization/sl/PrusaSlicer.po
 /usr/resources/localization/sl/PrusaSlicer.mo
 /usr/resources/localization/ko_KR/PrusaSlicer_ko.po
 /usr/resources/localization/ko_KR/PrusaSlicer_ko_KR.po
 /usr/resources/localization/ko_KR/PrusaSlicer_ko_KR.mo
 /usr/resources/localization/ko_KR/PrusaSlicer.mo
 /usr/resources/localization/it/PrusaSlicer_it.po
 /usr/resources/localization/it/PrusaSlicer.mo
 /usr/resources/localization/fi/PrusaSlicer.mo
 /usr/resources/localization/fi/PrusaSlicer_fi.po
 /usr/resources/localization/pt_BR/PrusaSlicer_pt_BR.po
 /usr/resources/localization/pt_BR/PrusaSlicer.mo
 /usr/resources/localization/ca/PrusaSlicer.mo
 /usr/resources/localization/ca/PrusaSlicer_ca.po
 /usr/resources/localization/wx_locale/hr.po
 /usr/resources/localization/wx_locale/ko.po
 /usr/resources/localization/wx_locale/pt_BR.po
 /usr/resources/localization/wx_locale/ar.po
 /usr/resources/localization/wx_locale/es.po
 /usr/resources/localization/wx_localeru.po
 /usr/resources/localization/wx_locale/nb.po
 /usr/resources/localization/wx_locale/pt.po
 /usr/resources/localization/wx_locale/ko_KR.po
 /usr/resources/localization/wx_locale/uk.po
 /usr/resources/localization/wx_locale/fr.po
 /usr/resources/localization/wx_locale/zh_CN.po
 /usr/resources/localization/wx_localero.po
 /usr/resources/localization/wx_locale/en.mo
 /usr/resources/localization/wx_locale/gl_ES.po
 /usr/resources/localization/wx_locale/lt.po
 /usr/resources/localization/wx_locale/de.po
 /usr/resources/localization/wx_locale/hi.po
 /usr/resources/localization/wx_locale/af.po
 /usr/resources/localization/wx_locale/fi.po
 /usr/resources/localization/wx_locale/sv.po
 /usr/resources/localization/wx_locale/eu.po
 /usr/resources/localization/wx_locale/sl.po
 /usr/resources/localization/wx_locale/be.po
 /usr/resources/localization/wx_locale/ne.po
 /usr/resources/localization/wx_locale/sk.po
 /usr/resources/localization/wx_locale/id.po
 /usr/resources/localization/wx_locale/tr.po
 /usr/resources/localization/wx_locale/lv.po
 /usr/resources/localization/wx_locale/fa_IR.po
 /usr/resources/localization/wx_locale/ja.po
 /usr/resources/localization/wx_locale/hu.po
 /usr/resources/localization/wx_locale/an.po
 /usr/resources/localization/wx_locale/ca@valencia.po
 /usr/resources/localization/wx_locale/ms.po
 /usr/resources/localization/wx_locale/en.po
 /usr/resources/localization/wx_locale/zh_TW.po
 /usr/resources/localization/wx_locale/vi.po
 /usr/resources/localization/wx_locale/sq.po
 /usr/resources/localization/wx_locale/pl.po
 /usr/resources/localization/wx_locale/cs.po
 /usr/resources/localization/wx_locale/el.po
 /usr/resources/localization/wx_locale/ca.po
 /usr/resources/localization/wx_locale/ta.po
 /usr/resources/localization/wx_locale/it.po
 /usr/resources/localization/wx_locale/nl.po
 /usr/resources/localization/wx_locale/da.po
 /usr/resources/localization/zh_CN/PrusaSlicer_zh_CN.po
 /usr/resources/localization/zh_CN/PrusaSlicer.mo
 /usr/resources/localization/de/PrusaSlicer_de.po
 /usr/resources/localization/de/PrusaSlicer.mo
 /usr/resources/localization/es/PrusaSlicer_es.po
 /usr/resources/localization/es/PrusaSlicer.mo
 /usr/resources/localization/be/PrusaSlicer_be.po
 /usr/resources/localization/be/PrusaSlicer.mo
 /usr/resources/localization/en/PrusaSlicer_en.po
 /usr/resources/localization/en/PrusaSlicer.mo
 /usr/resources/localization/zh_TW/PrusaSlicer_zh_TW.po
 /usr/resources/localization/zh_TW/PrusaSlicer.mo
 /usr/resources/localization/nl/PrusaSlicer_nl.po
 /usr/resources/localization/nl/PrusaSlicer.mo
 /usr/resources/localization/tr/PrusaSlicer_tr.po
 /usr/resources/localization/tr/PrusaSlicer.mo
 /usr/resources/localization/uk/PrusaSlicer_uk.po
 /usr/resources/localization/uk/PrusaSlicer.mo
 /usr/resources/localization/ru/PrusaSlicer_ru.po
 /usr/resources/localization/ru/PrusaSlicer.mo
 /usr/resources/localization/fr/PrusaSlicer_fr.po
 /usr/resources/localization/fr/PrusaSlicer.mo
 /usr/resources/localization/cs/PrusaSlicer_cs.po
 /usr/resources/localization/cs/PrusaSlicer.mo
 /usr/resources/localization/pl/PrusaSlicer_pl.po
 /usr/resources/localization/pl/PrusaSlicer.mo
 /usr/resources/localization/hu/PrusaSlicer_hu.po
 /usr/resources/localization/hu/PrusaSlicer.mo
```

For reasons unbeknownst to me, they are using `/usr/resources` rather than the more standard `/usr/share`.

The `resources` directory then gets copied into the AppDir. Maybe we need some environment variable to tell PrusaSlicer to load this from a relative (run-time) rather than an absolute (compile-time) path?

Samueru-sama, 2024-12-02:
> For reasons unbeknownst to me, Prusa are using /usr/resources rather than the more standard /usr/share.
> The `resources` directory then gets copied into the AppDir. Maybe we need some environment variable to tell PrusaSlicer to load this from a relative (run-time) rather than an absolute (compile-time) path?

Looks like the perfect scenario for this [library](https://github.com/fritzw/ld-preload-open).

```
PATH_MAPPING="/usr/resources:${SHARUN_DIR}/resources"
LD_PRELOAD=${SHARUN_DIR}/path-mapping.so
```

probonopd, 2024-12-02:
Let's see what @madpipeline responds regarding https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2512654819. Then we can retest in an environment similar to his.

novaplusplus, 2024-12-03:
Thanks for keeping this alive and showing that it's perfectly achieveable. Haven't used PS in a while but just grabbed your build to test while I try Nobara linux (light fork of Fedora 40) + KDE, seems to be working fine. Only just looked through basic functions though, did not go through and slice anything. Infamous web interface seems to work fine, as does the new printables integration

madpipeline, 2024-12-03:
> @madpipeline [...] Which language is your system set to?

```
env | grep -E 'LANG|LC_|CHARSET|LINGUAS'
LANG=ro_RO.UTF-8
LANGUAGE=

cat /etc/os-release
NAME="Fedora Linux"
VERSION="40 (KDE Plasma)"
ID=fedora
VERSION_ID=40
VERSION_CODENAME=""
PLATFORM_ID="platform:f40"
PRETTY_NAME="Fedora Linux 40 (KDE Plasma)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:40"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f40/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=40
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=40
SUPPORT_END=2025-05-13
VARIANT="KDE Plasma"
VARIANT_ID=kde
````

probonopd, 2024-12-03:
Thanks @madpipeline I can reproduce the issue and am working on an AppImage that fixes it.

probonopd, 2024-12-04:
@madpipeline please retry with https://github.com/probonopd/PrusaSlicer/releases/tag/mk2.
The difference is that this also bundles the `locales-all` package. I have tested it on Fedora Linux 40 (KDE Plasma) with `LANG=ro_RO.UTF-8`.

madpipeline, 2024-12-05:
It works ok. Thank you.

probonopd, 2024-12-20:
New build will be for 2.9.0-rc1

probonopd, 2024-12-24:
@SorrelArticulata thanks for reporting. Which operating system and version are you on?

rdragonrydr, 2024-12-26:
I recently updated using the new Flatpak system... I'm fine with that, but just to add/confirm to the discussion I did have to download the Gnome runtimes and that _is_ using some extra space. I believe it was 300M for me.

**I have a question related to this move, also** - How do I create a "personal" (unsigned) deb, appimage, or flatpak file (assuming the latter is possible without creating my own Flathub store) to compile from source? Currently, the source instructions create a large set of directories of mixed source and build files, and especially the part where it can't be moved afterward is inconvenient and somewhat messy with regards to my folders. I also can't copy it to another computer since if the user differs, the path will change.

If you can share the build script(s) for making a packaged version of the software (assuming a reasonably consistent or popular OS, or that we've figured out what dependencies to get installed from our package managers) so that it can be built from source but all self-contained and transferrable between computers, I'd really appreciate it.

Samueru-sama, 2024-12-26:
@rdragonrydr If I get this right you want to be able to transfer cloned source between devices and be able to each of those? I don't think that's possible unless you setup docker or similar, and still wouldn't be sure if you could move the cloned repos between devices either and more like run the build script on each PC instead.

You can see the steps to make the AppImage [here](https://github.com/probonopd/PrusaSlicer/blob/continuous/.github/workflows/build_appimage.yml#L76-L201).

iirc the CI builds several dependencies as well.

Worth noting that this appimage explicitly works on any linux distribution so that you can avoid having to do any building, also note that if you make a directory next to the appimage with the name of the appimage + `.home` the appimage runtime will automatically set that as `$HOME` so that you can also carry all the configuration files along with the app between devices.

rdragonrydr, 2024-12-26:
@Samueru-sama Not the cloned source, but I'd like to be able to personally build a binary and transfer _that_. If you check the latest release there _is_ no appimage anymore. I'd like to build my own from source (on one machine) and transfer that, or some other portable binary to other computers or to my quick-install scripts. It being usually located in the build folder and location locked is a problem for that plan.

Would running the steps from the link be enough to build an appimage, or would I need a VM or fakeroot or similar so it doesn't install itself in strange places as a side effect of the build? I'd like to be able to run subsequent builds without it clobbering something in my system...

Samueru-sama, 2024-12-26:
> @Samueru-sama Not the cloned source, but I'd like to be able to personally build a binary and transfer _that_. If you check the latest release there _is_ no appimage anymore. I'd like to build my [...]

You can get the AppImage here: https://github.com/probonopd/PrusaSlicer/releases/tag/continuous

This issue was opened precisely because Prusa stopped making the AppImage.

CRCinAU, 2024-12-27:
Seems like in Fedora 41, I get the following with the latest AppImage:

```
/tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_2.36' not found (required by /lib64/libdl.so.2)
/tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_ABI_DT_RELR' not found (required by /lib64/libdl.so.2)
/tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_2.36' not found (required by /lib64/librt.so.1)
/tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_ABI_DT_RELR' not found (required by /lib64/librt.so.1)
```

Samueru-sama, 2024-12-27:
> Seems like in Fedora 41, I get the following with the latest AppImage:
> 
> ```
> /tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_2.36' not found (required by /lib64/libdl.so.2)
> /tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_ABI_DT_RELR' not found (required by /lib64/libdl.so.2)
> /tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_2.36' not found (required by /lib64/librt.so.1)
> /tmp/.mount_PrusaSfIdDCO/shared/bin/prusa-slicer: /tmp/.mount_PrusaSfIdDCO/shared/lib/libc.so.6: version `GLIBC_ABI_DT_RELR' not found (required by /lib64/librt.so.1)
> ```
> 
> Fedora 41 has:
> 
> ```
> # rpm -qa | grep glibc | sort
> glibc-2.40-17.fc41.i686
> glibc-2.40-17.fc41.x86_64
> glibc-all-langpacks-2.40-17.fc41.x86_64
> glibc-common-2.40-17.fc41.x86_64
> glibc-devel-2.40-17.fc41.x86_64
> glibc-gconv-extra-2.40-17.fc41.i686
> glibc-gconv-extra-2.40-17.fc41.x86_64
> glibc-langpack-en-2.40-17.fc41.x86_64
> ```

Just booted a Fedora41 live image to test: 

[image]

-------------------------------------------------------------------------------------------------

Do you by chance have `LD_LIBRARY_PATH` or `LD_PRELOAD` set? 

On the terminal run `echo "$LD_LIBRARY_PATH" "$LD_PRELOAD"`

Because the AppImage already ships its own glibc and works on non-glibc systems as well, the only way I see you can get those glibc version errors is if something is making the AppImage load the host libraries.

CRCinAU, 2024-12-27:
```
echo "$LD_LIBRARY_PATH" "$LD_PRELOAD"
 /usr/NX/lib/libnxegl.so
```

Looks like that's for NoMachine remote desktop stuff.

Samueru-sama, 2024-12-27:
> ```
> echo "$LD_LIBRARY_PATH" "$LD_PRELOAD"
>  /usr/NX/lib/libnxegl.so
> ```
> 
> Looks like that's for NoMachine remote desktop stuff.

Now run 

```
unset LD_LIBRARY_PATH LD_PRELOAD
./PrusaSlicer-2.9.0-rc1-x86_64.AppImage 
```

It should work now.

CRCinAU, 2024-12-27:
Ahhhh - yep - that works... Is there a way to blank out those variables in the AppImage start? I can't see a valid use for them when libraries are bundled :|

Samueru-sama, 2024-12-27:
> Ahhhh - yep - that works... Is there a way to blank out those variables in the AppImage start? I can't see a valid use for them when libraries are bundled :|

First, how did you end up with `LD_PRELOAD` being set? Do you have that set in your `.bashrc` or similar?

https://www.hpc.dtu.dk/?page_id=1180 (Same applies for `LD_PRELOAD`). 

Modern appimage tooling actually does not make sure of `LD_LIBRARY_PATH` for these reasons, because if set then everything being launched by the AppImage will inherited those variables and that can cause these issues and they can be hard to notice why. 

Instead if you want to force a certain library, you should use `patchelf --add-rpath` `patchelf --add-needed` on the application/library that you need this. Or at the very least only use those variables when calling that specific application that needs them.

CRCinAU, 2024-12-27:
It looks like installing NoMachine drops in a file at `/etc/xdg/plasma-workspace/env/nx-sourceenv.sh` that contains:

```bash
export LD_PRELOAD=/usr/NX/lib/libnxegl.so
```

Samueru-sama, 2024-12-27:
> It looks like installing NoMachine drops in a file at `/etc/xdg/plasma-workspace/env/nx-sourceenv.sh` that contains:
> 
> ```shell
> export LD_PRELOAD=/usr/NX/lib/libnxegl.so
> ```

That's crazy, I was going to write a long rant about that but that won't change anything,

I forked the thing and applied [this fix](https://github.com/Samueru-sama/PrusaSlicer/commit/04c9338bceb0ea9f552658a340d57aba8fa45481)

The CI is building the AppImage but I think it will take a while to be built, check the releases in about 1 hour on my fork and confirm that it works. 

-------------------------------------------------------------------------------------

Btw I need to confirmation because I'm not 100% sure that what I did will work, because the AppImage does not even make use of the host shell to be setting/removing env variables, instead `sharun` uses [this](https://crates.io/crates/dotenv) to handle env variables and maybe `unset` isn't a valid term and I need to change that to `LD_PRELOAD=` (set it empty) for example.

Samueru-sama, 2024-12-27:
@CRCinAU Just tested it and it works. made a PR unsetting the variables you can either wait for it to be merged or use the one at [my fork](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous)

CRCinAU, 2024-12-28:
> @CRCinAU Just tested it and it works. made a PR unsetting the variables you can either wait for it to be merged or use the one at [my fork](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous)

I actually went a slightly different route and removed that file installed by NoMachine - as I'm not exactly sure its required.

NoMachine seems to work fine without it - however, its good to have this fix in place anyway :)

jseyfert3, 2025-01-01:
@probonopd I downloaded the AppImage from the linked repo from 4 days ago and ran it. It booted fine, I imported a .step file and got the following error:
> **PrusaSlicer has encountered an error**
> Cannot load OCCTWrapper.so:
>/tmp/.mount_PrusaShmAjoE/bin/OCCTWrapper.so: cannot open shared object file: No such file or directory

Linux Mint 22

I then tried importing a .stl file, and it imported fine, and sliced fine. But when I tried to rotate the sliced object, I heard the fans on my computer kick into high gear, and rotating was super jerky. Continuous movement lead to some 25% CPU usage, which explains the fans kicking up. I have a pretty beefy computer too:

- AMD Ryzen 5 7600 X
- 64 GB DDR5, in two 32 GB sticks
- Samsung 980 PRO 1 TB M.2 SSD
- Geforce Nvidia GTX 1080 
- 3440x1440 monitor running at 165 Hz

I then fired up PrusaSlicer 2.9.0 from the Flatpak, loaded the same .stl, and sliced. It seemed to slice faster. After slicing, I rotated it continuously like before. It only took some 1-2% of CPU usage and the rotate was butter smooth.

Oh, I just figured it out rotating it again with Psensor running. While rotating, my GPU usage goes up to ~30%. The AppImage you made isn't using my GPU.

I verified the last statement by firing up the Prusaslicer 2.8.1 x64 GTK3 appimage from the Prusa release. Just like  PrusaSlicer 2.9.0 from the Flatpak, rotating the sliced model uses about ~30% of my GPU and 1-2% of my CPU.

Summary of what little initial testing I've done:

1. Importing a .step file fails with an error message about not being able to locate OCCTWrapper.so
2. PrusaSlicer is not using my GPU in your AppImage, but does in the 2.9.0 FlatPak _and_ in Prusa's 2.8.1 AppImage.

Samueru-sama, 2025-01-01:
The `OCCTWrapper.so` error is because the lib was never bundled on the AppImage, on the older AppImage this library was placed on `bin` next to the prusa binary.

This library is also likely only dlopened when you try to open such files, hence it was never caught by sharun when deploying. 

`OCCTWrapper.so` would need to be added to the top level `bin` directory in the AppDir.

---------------------------------------------------

And the high cpu usage is likely because the proprietary nvidia driver doesn't work with the AppImage, the AppImage already bundles the `nouveau` libs but you have a 10 series nvidia GPU which support is basically non existent as far as I know. The older AppImage instead uses the host gpu drivers which is why it works in that case.

Samueru-sama, 2025-01-01:
@jseyfert3 Please try [this](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous) and let me know.

CRCinAU, 2025-01-01:
As we're looking at feature parity, I noticed that when I sliced and sent my job to the printer, mdns didn't seem to work to send the job to my printer at `ender3.local` - however, changing it to IP address works fine.

jseyfert3, 2025-01-01:
> And the high cpu usage is likely because the proprietary nvidia driver doesn't work with the AppImage, the AppImage already bundles the nouveau libs but you have a 10 series nvidia GPU which support [...]
Ah. That would probably explain it. I'm not running nouveau drivers on the host OS, I'm running the Nvidia driver, specifically 535.183.01-0ubuntu0.24.04.1

I know you can't bundle an Nvidia driver into an open-source package though, since it's not an open-source driver.

jseyfert3, 2025-01-01:
> @jseyfert3 Please try [this](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous) and let me know.
@Samueru-sama loads .step files now. Doesn't change the slow sliced part rotation though. I would not want to use the AppImage purely because of the slowness of the graphical performance. It's much worse than using it on a laptop with integrated graphics, which is what I do occasionally when I have to print a part at work. But in that case, my work laptop is running Windows so an AppImage isn't in the question.

I'm half curious to try rebooting and setting BIOS to use integrated graphics on my desktop, but of course as a user I'd never want to have to disable my GPU just to make the program work well.

Samueru-sama, 2025-01-01:
>As we're looking at feature parity, I noticed that when I sliced and sent my job to the printer, mdns didn't seem to work to send the job to my printer at ender3.local - however, changing it to IP address works fine.

mmm here I have no idea why this would happen. 

>Ah. That would probably explain it. I'm not running nouveau drivers on the host OS, I'm running the Nvidia driver, specifically 535.183.01-0ubuntu0.24.04.1

You don't have to be running nouveau drivers on the host in order to use them, the AppImage already has them and will try to use them since it ships them and sets `__EGL_VENDOR_LIBRARY_DIR` and `LIBGL_DRIVERS_PATH`. 

The reason is falls back to using the cpu is likely because the nouveau drivers just have poor performance/don't work at all with 10 series gpu, iirc not long ago they recently figured out how to make the gpu clock higher on 10 series to give you an idea on how progress is going for pascal. 

>I'm half curious to try rebooting and setting BIOS to use integrated graphics on my desktop, but of course as a user I'd never want to have to disable my GPU just to make the program work well.

You can try to enable the igpu without setting it as the default gpu nor connec the the monitor to it.

CRCinAU, 2025-01-01:
> mmm here I have no idea why this would happen.

I'm guessing the `/etc/nsswitch.conf` file doesn't have mdns enabled - assuming it has one. 

It should have a line like:
```
hosts:      files myhostname mdns_minimal [NOTFOUND=return] resolve [!UNAVAIL=return] nis dns
```

It also needs the nss-mdns package.

Samueru-sama, 2025-01-01:
@CRCinAU [try this](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous)

CRCinAU, 2025-01-01:
> @CRCinAU [try this](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous)

Hmmm - there might be something else required....

```
[2025-01-01 17:22:01.248685] [0x00007f0a410667c0] [error]   OctoPrint: Error getting version: Couldn't resolve host name:
Could not resolve host: ender3.local
[Error 6], HTTP 0, body: ``
[2025-01-01 17:22:30.447931] [0x00007f0a410667c0] [error]   OctoPrint: Error getting version: Couldn't resolve host name:
Could not resolve host: ender3.local
[Error 6], HTTP 0, body: ``
```

However, from the system shell:
```
$ ping ender3.local
PING ender3.local (fe80::37c0:7aa6:b633:99cb%enp36s0) 56 data bytes
64 bytes from ender3.local (fe80::37c0:7aa6:b633:99cb%enp36s0): icmp_seq=1 ttl=64 time=0.356 ms
64 bytes from ender3.local (fe80::37c0:7aa6:b633:99cb%enp36s0): icmp_seq=2 ttl=64 time=0.148 ms
64 bytes from ender3.local (fe80::37c0:7aa6:b633:99cb%enp36s0): icmp_seq=3 ttl=64 time=0.104 ms
^C
--- ender3.local ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2089ms
rtt min/avg/max/mdev = 0.104/0.202/0.356/0.109 ms
$ ping ender3.local -4
PING ender3.local (172.31.1.5) 56(84) bytes of data.
64 bytes from 172.31.1.5: icmp_seq=1 ttl=64 time=0.186 ms
64 bytes from 172.31.1.5: icmp_seq=2 ttl=64 time=0.141 ms
^C
--- ender3.local ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1002ms
rtt min/avg/max/mdev = 0.141/0.163/0.186/0.022 ms
```

jseyfert3, 2025-01-01:
>You don't have to be running nouveau drivers on the host in order to use them, the AppImage already has them and will try to use them since it ships them and sets __EGL_VENDOR_LIBRARY_DIR and LIBGL_DRIVERS_PATH.

Yeah, I get that. I mean that explains why the one you made, that is using nouveau drivers, is running so poorly.

>The reason is falls back to using the cpu is likely because the nouveau drivers just have poor performance/don't work at all with 10 series gpu, iirc not long ago they recently figured out how to make the gpu clock higher on 10 series to give you an idea on how progress is going for pascal.

I tested this by switching my system drivers to nouveau. The 2.8.1 AppImage suddenly behaved as poorly as the one you made. Wow, the nouveau drivers are terrible for this series Nvidea CPU.

>You can try to enable the igpu without setting it as the default gpu nor connect the the monitor to it.

I tried. It was frustrating on many levels. Mainly because I have an odd combination of a GPU that's fairly old and a fairly new, high res OLED monitor. I can't view POST screen on the monitor, nor the Mint boot screens. It's not until the system boots to user login that the OLED monitor recognizes a video stream and powers on. I forgot the reason, but the OLED monitor is expecting something that you don't get in POST or OS boot until the Nvidia 2000 series and onwards GPUs, I think. I forget what it is. My old monitor can view POST and Mint boot screens, but it's DVI, and my GPU does have a DVI port but if I leave a DisplayPort cable and monitor connected it won't use the DVI port.

Anyway, bottom line was most I could do was get Mint to display both NVidea and AMD integrated on the System Info box, but performance was bad. I feel it was still using the nouveau drivers and the NVidea card. I could try removing the GPU entirely, to force it to integrated and only integrated graphics, but that's going a bit far. (I tried setting BIOS to force integrated graphics, but it didn't disable the NVidea card, seemingly)

Is there a reason to not just use the host OS graphics drivers, as Prusa does?

Samueru-sama, 2025-01-01:
@CRCinAU [Try again](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous) in abou 5 minutes since it turns out the ubuntu runner didn't have the mdns libs. 

If it fails I need to start reading about the whole mdns thingy, this is actually part of glibc the `/etc/nsswitch.conf` is read by `libc.so.6` and from there it determines to use the mdns libs. 

-------------------------------------------------------------------

@jseyfert3 Just in case, at no point you have to connect the igpu to a display, just have it enable, you can totally use the igpu to do hardware acceleration while displaying the contents using another gpu, but seems it didn't work anyway. 

>Is there a reason to not just use the host OS graphics drivers, as Prusa does?

This AppImage bundles all its dependencies and doesn't use the ones from the host, that includes the gpu drivers. 

Now this is to allow running the AppImage on more systems than what normally appimages cover, which is basically all GNU/Linux systems equal or newer than ubuntu 20.04. This AppImage instead actually runs even on musl linux systems like alpine and non FHS systems like NixOS, sharun doesn't even use the host shell to set env variables either. 

The original issue Prusa had was had to do with webkitgtk, to be honest that original issue could have been fixed without doing these other changes, but it seems it was also a common complain that the AppImage really didn't work on all linux systems so this change was made. 

The issue with webkitgtk [is this part](https://github.com/Samueru-sama/PrusaSlicer/blob/continuous/.github/workflows/build_appimage.yml#L130-L142)

So to answer the question, it cannot be done, however there is something you could try as last shot, but I need to know the contents of all the files you have in `/usr/share/glvnd`. You can send them here instead of pasting the entire content.

CRCinAU, 2025-01-01:
> @CRCinAU [Try again](https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous) since it turns out the ubuntu runner didn't have the mdns libs.
> 
> If it fails I need to start reading about the whole mdns thingy, this is actually part of glibc the `/etc/nsswitch.conf` is read by `libc.so.6` and from there it determines to use the mdns libs.

[image]

Perfect - thanks :)

probonopd, 2025-01-01:
Thank you very much @Samueru-sama. I have merged your improvements in my build.

Do you know a way to force software rendering instead of using Nouveau? I've never had positive experiences with the performance of Nouveau in the past.

@jseyfert3: Possibly a good software renderer might be more performant than Nouveau on your system? I wonder what would happen if you would run in a terminal

```
export MESA_LOADER_DRIVER_OVERRIDE=llvmpipe
export GALLIUM_DRIVER=llvmpipe
export LIBGL_ALWAYS_INDIRECT=0
```

and then execute PrusaSlicer from the same terminal.

jseyfert3, 2025-01-01:
> This AppImage bundles all its dependencies and doesn't use the ones from the host, that includes the gpu drivers.
>
> Now this is to allow running the AppImage on more systems than what normally [...]

@Samueru-sama it sounds like it can be done, but in the interest of trying to made it 100% self contained, you are choosing not to do it. Because it used to be done, and FreeCAD does it too. And the Flatpak version of PrusaSlicer does too? How does the Flatpak version of PrusaSlicer do this, but the AppImage "cannot"?

There's only one directory in /usr/share/glvnd, egl_vendor.d, containing 10_nvidia.json:
```                                                               
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
```
and 50_mesa.json:
```
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_mesa.so.0"
    }
}
```

jseyfert3, 2025-01-01:
>@jseyfert3: Possibly a good software renderer might be more performant than Nouveau on your system? I wonder what would happen if you would run in a terminal
>
>export MESA_LOADER_DRIVER_OVERRIDE=llvmpipe
export GALLIUM_DRIVER=llvmpipe
export LIBGL_ALWAYS_INDIRECT=0
>
>and then execute PrusaSlicer from the same terminal.

@probonopd 
```
sasquatch@sasquatch-A620I-AX:~/Apps$ export MESA_LOADER_DRIVER_OVERRIDE=llvmpipe
export GALLIUM_DRIVER=llvmpipe
export LIBGL_ALWAYS_INDIRECT=0
sasquatch@sasquatch-A620I-AX:~/Apps$ ./PrusaSlicer-2.9.0-rc1-x86_64.AppImage 
Gtk-Message: 11:25:15.947: Failed to load module "xapp-gtk3-module"
[2025-01-01 11:25:16.810287] [0x00007d94677797c0] [error]   UserAccount: Failed to read token - no datafile found.
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it
11:25:17: Debug: window wxTreeCtrl@0x555577acf730 ("treeCtrl") lost focus even though it didn't have it

** (prusa-slicer:5045): WARNING **: 11:25:42.169: atk-bridge: get_device_events_reply: unknown signature
sasquatch@sasquatch-A620I-AX:~/Apps$ 
```
Performance seemed the same as just launching the AppImage by double-clicking it from the file explorer.

CRCinAU, 2025-01-01:
> sasquatch@sasquatch-A620I-AX:~/Apps$ export MESA_LOADER_DRIVER_OVERRIDE=llvmpipe
> export GALLIUM_DRIVER=llvmpipe
> export LIBGL_ALWAYS_INDIRECT=0
> sasquatch@sasquatch-A620I-AX:~/Apps$ ./PrusaSlicer-2.9.0-rc1-x86_64.AppImage

That may not of worked... I'd expect a shell prompt back after every `export` command.

jseyfert3, 2025-01-01:
>That may not of worked... I'd expect a shell prompt back after every export command.

I pasted as three lines at once. I can try doing them one at a time.

Addendum to the above: The same messages appear when launching the AppImage in a new terminal window that I had not done the `export` commands on.

Samueru-sama, 2025-01-01:
>Addendum to the above: The same messages appear when launching the AppImage in a new terminal window that I had not done the export commands on.

Is the performance the same after you run:

 `export MESA_LOADER_DRIVER_OVERRIDE=llvmpipe GALLIUM_DRIVER=llvmpipe LIBGL_ALWAYS_INDIRECT=0` 
 
 And then run the AppImage from the terminal? Note that you can run a single export setting the 3 variables instead of doing it one by one.
 
If performance is still the same the next thing to try is to force usage of the proprietary nvidia driver, run the appimage with `--appimage-extract` and on the `squashfs-root` directory go to `usr/share/glvnd/egl_vendor.d` and just copy and paste your `10_nvidia.json` and the line that says `"library_path" : "libEGL_nvidia.so.0"` add the full path to where you have `libEGL_nvidia.so.0` on your system, iirc it is `/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0` on linux mint.

This will most likely fail but it is worth trying.

jseyfert3, 2025-01-01:
Addendum 2: Export commands do work. There's a noticeable difference in GPU memory, usage %, and CPU usage % running the AppImage with and without the export commands. Performance is bad in either case.
[image]

Here's a quick shot of constant rotation in the 2.8.1 AppImage. ~30% GPU usage, ~7% GPU memory usage, almost no CPU usage. You can see the earlier runs with the 2.9.0-rc AppImage further to the left in the Psensor graph. Interesting that the 2.9.0-rc AppImage uses considerably more GPU memory.
[image]

Also, just for the sake of comparison, the messages from launching and slicing an object in 2.8.1 AppImage:
```
sasquatch@sasquatch-A620I-AX:~/Apps$ ./PrusaSlicer-2.8.1+linux-x64-newer-distros-GTK3-202409181416.AppImage 

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.959: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.959: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.960: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.960: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.961: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.961: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.961: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.961: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.962: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.962: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.962: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.962: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.964: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.964: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:39.965: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.965: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.972: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.972: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.972: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.972: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.973: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.973: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.973: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:39.973: gtk_widget_set_size_request: assertion 'width >= -1' failed
[2025-01-01 11:52:40.008267] [0x0000711a0ec53080] [error]   UserAccount: Failed to read token - no datafile found.

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:40.084: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:40.129: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:40.129: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:40.130: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:40.130: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:40.130: invalid cast from 'wxPizza' to 'GtkCellLayout'
[... 101 lines cut ...]
11:52:42: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it
[... cut ...]
11:52:43: Debug: window wxTreeCtrl@0x6068ddc432c0 ("treeCtrl") lost focus even though it didn't have it

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:50.446: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:6883): GLib-GObject-CRITICAL **: 11:52:53.273: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:6883): Gtk-CRITICAL **: 11:52:53.273: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
sasquatch@sasquatch-A620I-AX:~/Apps$ 
```

jseyfert3, 2025-01-01:
>If performance is still the same the next thing to try is to force usage of the proprietary nvidia driver, run the appimage with --appimage-extract and on the squashfs-root directory go to usr/share/glvnd/egl_vendor.d and just copy and paste your 10_nvidia.json and the line that says "library_path" : "libEGL_nvidia.so.0" add the full path to where you have libEGL_nvidia.so.0 on your system, iirc it is /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0 on linux mint.
>
>EDIT: After that's done run the AppRun in squashfs-root
>
>This will most likely fail but it is worth trying.

I copied it and changed it to:
```
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0"
    }
}
```
after verifying that library does exist at that location.
```
sasquatch@sasquatch-A620I-AX:~/Apps$ ./PrusaSlicer-2.9.0-rc1-x86_64.AppImage --appimage-extract
sasquatch@sasquatch-A620I-AX:~/Apps$ cd squashfs-root/
sasquatch@sasquatch-A620I-AX:~/Apps/squashfs-root$ ls
AppRun  bin  etc  lib  PrusaSlicer.desktop  PrusaSlicer.png  resources  share  shared  sharun  usr
sasquatch@sasquatch-A620I-AX:~/Apps/squashfs-root$ ./AppRun
Gtk-Message: 12:14:37.771: Failed to load module "xapp-gtk3-module"
[2025-01-01 12:14:38.474790] [0x00007970cc7587c0] [error]   UserAccount: Failed to read token - no datafile found.
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it
12:14:39: Debug: window wxTreeCtrl@0x555576813950 ("treeCtrl") lost focus even though it didn't have it

** (prusa-slicer:7935): WARNING **: 12:15:03.727: atk-bridge: get_device_events_reply: unknown signature
sasquatch@sasquatch-A620I-AX:~/Apps/squashfs-root$
```
I ran it, and it did not use my GPU.

jseyfert3, 2025-01-01:
I guess the thing that confuses me is, if Flatpak bundles "everything" like Prusa's 2.9.0 release notes say, then why can the Flatpak version of PrusaSlicer use my Nvidia GPU, apparently with the Nvidia proprietary drivers that by definition cannot be bundled in an open source package, but the AppImage cannot?

jseyfert3, 2025-01-01:
Huh. Maybe this answers my question.
[image]

jseyfert3, 2025-01-01:
Yeah, it did. So Flatpak works by just downloading the Nvidia proprietary drivers separately from the host OS drivers. I forgot that yesterday when testing switching my system to nouveau, I switched to the nvidia-550 driver when I went back.

I tried `flatpak list --runtime' and no nvidia results were listed, and the Mint Update Manager was saying there was the nvidia 550 driver available. Out of curiosity I ran the Flatpak PS and...no 3D display, just the rest of the program:
[image]

Installed the flatpak nvidia-550 runtime package and:
[image]

So that answers that. You can't bundle Nvidia drivers, so Flatpak just installs Nvidia drivers separately, if an app wants them. And for AppImage, since everything is supposed to be in one file, and you can't include the Nvidia drivers, your only option is to either use the super poor performance of bundled nouveau drivers, or to use whatever drivers the host OS has installed? The former meaning the performance is terrible for people like me running at least certain nvidia GPUs, and the latter meaning the AppImage has restricted portability, right?

Samueru-sama, 2025-01-01:
Yes flatpak downloads the proprietary nvidia driver, doesn't use the one from the host. 

Something similar can be done with the AppImage, [conty](https://github.com/Kron4ek/Conty) has it and we use it with [this AppImage](https://github.com/ivan-hc/Steam-appimage) for example. [conty actually pulls the nvidia driver from flathub](https://github.com/Kron4ek/Conty/blob/master/conty-start.sh#L409).

But this is going to be a lot of work to adapt this AppImage to use that method, and also some studying needs to be done because unlike conty/flatpak this AppImage does not make use of containers.

>The former meaning the performance is terrible for people like me running at least certain nvidia GPUs, and the latter meaning the AppImage has restricted portability, right?

Yes. But once again to be honest the original issue Prusa ran into had to do with webkitgtk, that could have been fixed using the old method the AppImage was being made with. Bundling everything like we did here is just an extra that improves the portability but had this unfortunate side effect.

jseyfert3, 2025-01-01:
>Yes. But once again to be honest the original issue Prusa ran into had to do with webkitgtk, that could have been fixed using the old method the AppImage was being made with. Bundling everything like we did here is just an extra that improves the portability but had this unfortunate side effect.

Yeah I get that now. How would this affect people with AMD GPUs? If I understand correctly AMD pushes their drivers to the kernel. Does that mean this AppImage would work great for people running AMD GPUs?

Samueru-sama, 2025-01-01:
> Yeah I get that now. How would this affect people with AMD GPUs? If I understand correctly AMD pushes their drivers to the kernel. Does that mean this AppImage would work great for people running AMD GPUs?

Yes hardware acceleration should be working on AMD, Intel and as far as I know nvidia 16 series and newer since that's where nouveau is good.

CRCinAU, 2025-01-01:
> Yes hardware acceleration should be working on AMD, Intel and as far as I know nvidia 16 series and newer since that's where nouveau is good.

I run an AMD 7600XT and I don't have any issues at all.

Samueru-sama, 2025-01-02:
Would be nice to get this tested on a 16 series nvidia gpu or newer to make sure `nouveau` is working.

Also @jseyfert3 you should totally be able to have the AMD igpu do the hardware acceleration and the dedicated gpu do the display and this should happen automatically but seems it doesn't work on your end 🤔

jseyfert3, 2025-01-04:
>Also @jseyfert3 you should totally be able to have the AMD igpu do the hardware acceleration and the dedicated gpu do the display and this should happen automatically but seems it doesn't work on your end 🤔

Well, that was highly annoying. Having a GPU that won't output video to my primary monitor during BIOS or boot makes it _very_ annoying to try changing graphics output settings...

Anyway, I eventually ended up with no GPU installed and the integrated graphics outputting. Result: Terrible performance on all 3 PrusaSlicers I ran, 40-50% CPU usage while rotating them. I took a look at PassMark's GPU benchmarks, and the 7600X integrated GPU has a benchmark of 1743. The 1080 has a benchmark of 15,555. So yeah, it probably was running on integrated GPU at one point and really couldn't tell a performance difference over the 1080 with nouveau drivers, because my integrated GPU _sucks_. Of course, that makes sense, because AMD didn't even used to put _any_ integrated GPU into the X version of their CPUs, since the X version is for performance oriented consumers, basically all of whom will have dedicated GPUs.

[71 more replies]