prusa3d/PrusaSlicer #13655: Paint on Support chrashes the Slicer with specific stl-File

U-B-Glade, 2024-11-25:
### Description of the bug

If I try to use paint on support, the Slicer freeze and closes itself.  
It occour even by using it automatic or manual.

Windows Event is:
Name der fehlerhaften Anwendung: prusa-slicer.exe, Version: 2.8.1.0, Zeitstempel: 0x66eaddda
Name des fehlerhaften Moduls: PrusaSlicer.dll, Version: 0.0.0.0, Zeitstempel: 0x66eaddd2
Ausnahmecode: 0xc0000005
Fehleroffset: 0x0000000000601f52

### Project file & How to reproduce

[FCH_LF_crash.zip](https://github.com/user-attachments/files/17905430/FCH_LF_crash.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

Windows 10 pro 10.0.19045

### Printer model

i3 MK3S+ MMU2S

Jan-Soustruznik, 2024-11-26:
Hello, @U-B-Glade, what do you mean by "Only occur on low RAM", thank you

U-B-Glade, 2024-11-26:
If I have 8 of 16 GB avaiable it works slow but it works, if In have less then 4GB it crashes, 
don't know why but it ist a 0x5, so no wonder :-)

closed 2026-09-29 by prusaslicer-bot[bot]