prusa3d/PrusaSlicer #13659: Include <chrono> for chrono_literals

NEIL-smtg, 2024-11-26:
### Description of the bug

I work on Microsoft Visual C++ testing, where we regularly build popular open-source projects, including yours, with development builds of our compiler and libraries to detect and prevent shipping regressions that would affect you. This also allows us to provide advance notice of breaking changes, which is the case here.

Recently https://github.com/microsoft/STL/pull/5105, revealed an issue in PrusaSlicer.

Compiler error with this STL change:
```c++
C:\gitP\prusa3d\PrusaSlicer\src\libslic3r\Support\TreeModelVolumes.cpp(169,33): error C3083: 'high_resolution_clock': the symbol to the left of a '::' must be a type [C:\gitP\prusa3d\PrusaSlicer\build_amd64\src\libslic3r\libslic3r.vcxproj]
C:\gitP\prusa3d\PrusaSlicer\src\libslic3r\Support\TreeModelVolumes.cpp(169,56): error C2039: 'now': is not a member of 'std::chrono' 
```

Affected code:
https://github.com/prusa3d/PrusaSlicer/blob/ae97d00c34258f85b98fb000c131162863424a04/src/libslic3r/Support/TreeModelVolumes.cpp#L169

This was assuming that including <thread> makes the chrono::high_resolution_clock type available, which is not guaranteed by the Standard. You must explicitly include <chrono> into this file and the error will be resolved.

### Project file & How to reproduce

Build log: [prusaSlicer.txt.log](https://github.com/user-attachments/files/17913642/prusaSlicer.txt.log)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

latest commit

### Operating system

windows 11

### Printer model

-

SachCZ, 2024-11-26:
Hi @NEIL-smtg. Makes sense, thank you.

SachCZ, 2024-11-26:
Hello, I take it back. It does not seem to make sense. There is `#include <chrono>` in the linked file.

NEIL-smtg, 2024-11-27:
My apologies, everyone. There is no compile error when using the latest commit of PrusaSlicer. Closing this issue now.

closed 2024-11-27 by NEIL-smtg