prusa3d/PrusaSlicer #13661: When using multiple beds, splitting to objects sometimes moves the now-split object

rcmaniac25, 2024-11-26:
### Description of the bug

Utilizing multi-bed feature, I had a model that was specifically broken into individual "plates" with multiple models for each one (combined into 1 model). I put each plate on it's own bed. I then split the model. Instead of the model simply being split, it moved the now-split model to another location.

### Project file & How to reproduce

1. Have a model with multiple parts and multiple plates (example: https://www.printables.com/model/1063554-chunky-venom-kit-no-support-no-ams-no-glue)
2. Arrange the plates of parts to multiple beds
3. Activate a different bed
4. Right click the plate of parts and choose "Split" -> "Split to Parts"

A combined plate of parts (before splitting)
[image]

Location of parts after splitting to objects
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-alpha1

### Operating system

Windows 10

### Printer model

Prusa MK4S+MMU3

Jan-Soustruznik, 2024-12-02:
Hello @rcmaniac25, Thank you for your report. I’ve created an internal ticket (SPE-2600). We’ll aim to address it in one of the upcoming releases.

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo