prusa3d/PrusaSlicer #13665: Unable to close "Send to Connect/Choose a printer" - window

Spreadcat, 2024-11-26:
### Description of the bug

In the window for choosing a printer when sending the gcode to prusa-connect, there's is no way/cancel button to abort the printer selection.
The workflow suggest to send the file to the printer and delete it then, if you want to abort the print.

[image]

### Project file & How to reproduce

1. Slice any stl file.
2. Click on the button "Send to Connect" in the bottom corner of the PrusaSlicer.
3. The window for the printer selection opens.
4. Attempt to abort the print selection.

[config](https://github.com/user-attachments/files/17918573/shark.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Fedora 38

### Printer model

MK4S

ballessay, 2025-03-02:
Well nagged by this I had a look in the source. I have to admit I'm not familiar with wxWidgets. I found out the follwing:

Since `class PrinterPickWebViewDialog` inherits from `class WebViewDialog` and that class does the [this](https://github.com/prusa3d/PrusaSlicer/blob/55139f493381b510716a7813564acfc2c353a267/src/slic3r/GUI/WebViewDialog.cpp#L139) in its constructor, it is possible to close the dialog by "close" keyboard shortcut of your OS or your specific window manager.

In my case, since "ctrl+w" and "ESC" weren't working I now have to resort to [kill the window in i3](https://i3wm.org/docs/userguide.html#_closing_windows).

I hope this helps to work around this.

Best regards

Spreadcat, 2025-03-03:
Hei,

> In my case, since "ctrl+w" and "ESC" weren't working I now have to resort to [kill the window in i3](https://i3wm.org/docs/userguide.html#_closing_windows).

Yes, I can confirm. That's the way  close it currently as well .. or as long as I encountered the problem. I don't expect users who are not familiar with i3 to be able to do the same. Quite frankly, I have no idea how on non-i3 desktops the window can be closed.

It looks to me that the current flow with "select/confirm" is not fully implemented and the cancel button (or a window x-button) just simply has been forgotten maybe.

ballessay, 2025-04-30:
Hello again...
> Yes, I can confirm. That's the way close it currently as well .. or as long as I encountered the problem. I don't expect users who are not familiar with i3 to be able to do the same. Quite frankly, I have no idea how on non-i3 desktops the window can be closed.

Nice that it works for you too. Haven't recognized i3 in your screenshot, to be honest. XD

I'd guess the dialog has windows decorations, like mini-/maximize and close buttons, on  desktop environments/windows managers. But that's just a guess.

Spreadcat, 2025-05-02:
That's a guess also from my side. Guess that leaves two questions then:

1. Is there a close/cancel button for non-i3 users?
2. Should there a button for i3 users as well (or for all if it's missing)?

closed 2026-09-29 by prusaslicer-bot[bot]