prusa3d/PrusaSlicer #13668: Exploded view for mutli-color

The-Original-Reth, 2024-11-26:
**Describe the solution you'd like**
I would like to see PrusaSlicer incorporate exploded view when doing multi-material prints. 

**Describe how it would work**
Currently Bamboo Slicer and Orca Slicer allow for an "assembly view". What this does is explodes a multi object/part model so that one can paint the individual files without having to deal with the paint brush. 
To my knowledge this works on a single file (obj, step, etc.) with multiple objects, or multiple files incorporated at a single time.

[image]

u89djt, 2024-11-26:
(Fellow user who recognizes that you can achieve what you need, though not precisely what you asked for there)
When you right click a single stl that contains multiple bodies intended for different colours, you can select "split to parts" and you can then set the tool for each part. The parts will stay in their original positions. Use the tool drop-down menu for each part in the objects/parts list.

u89djt, 2024-11-26:
When I loaded the set of stls into prusaslicer it gave me the option of placing them as parts of a single object, so they were already separately coloured.
[Turtle.zip](https://github.com/user-attachments/files/17925185/Turtle.zip)
[image]

u89djt, 2024-11-26:
Absolutely. Not what you asked for.

closed 2026-09-29 by prusaslicer-bot[bot]