prusa3d/PrusaSlicer #13669: 2.9 Alpha1 - Avoid crossing perimeters - better, but not fully fixed

marsmanp, 2024-11-26:
### Description of the bug

The same as seen here: https://github.com/prusa3d/PrusaSlicer/issues/13445

The Avoid crossing perimeters being enabled does not always force travels to be within the object even if it is possible to do so.  In the attached file you can get great/good/poor/bad results depending on other settings like number of perimeters and infill type.

### Project file & How to reproduce

[ACP.zip](https://github.com/user-attachments/files/17925715/ACP.zip)
A few examples of different settings for perimeters/infill causing changes in travels crossing/not crossing perimeters.
[image]
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0 alpha1

### Operating system

Windows 10

### Printer model

Anycubic MegaPro

SachCZ, 2024-11-26:
Hello @marsmanp, thank you for the report. We know about this and it will be fixed in the next iteration.

Jan-Soustruznik, 2024-11-27:
SPE-2578

dattas, 2025-02-12:
Just want to verify things since the other tickets got closed incorrectly as the issue persists. This is still a problem and is causing problems with my Prusa XL causing stringing by crossing the perimeters. Is this still being looked at improving further? I know it isn't just me as others are also indicating there is still an issue: https://github.com/prusa3d/PrusaSlicer/issues/12432#issuecomment-2568082778

pmarks-net, 2025-02-13:
Here is a test case where "Avoid crossing perimeters" fails: https://www.thingiverse.com/thing:6948297

### PrusaSlicer 2.9.0:

[image]

### Cura 5.9.0 for comparison:

[image]

SachCZ, 2025-02-26:
Hello, thank you for your feedback. I just checked and the **original** issue is fixed. When it comes to @pmarks-net example, I am not aware of PrusaSlicer ever working like cura in this case. It is a difference in the algorithm used. You can open a feature request if this is something you want us to implement.

The @dattas case is similar in that I do not **think this is a regression**. When moving to other objects (wipe tower including) the move is usually **direct**.
There is no way around it, for example, imagine you are going from object infill to a wipe tower. Because the **perimeter is a closed loop** you have to cross it one way or another. In that case it is better to just go directly.
The same **holds for the multi material islands.** Each has its own perimeters, meaning if you go from within the island elsewhere, you will have to cross it. Is is better to just go directly.

The direct move mentioned tries to avoid "holes" in the islands (**islands are single material, bounded by perimeters**), but in the example @dattas provided, there are no holes, because of the multi material setup, so nothing to really avoid when going from object to object. 

If you still believe there is a regression, please provide a PrusaSlicer version where this works as expected and version where it is broken.

If this is not a bug but something you would want to work, please open a feature request where you describe in detail how it should work.

Thank you for your time and effort.

closed 2025-02-26 by SachCZ